microsoft/vscode-arduino #468: Can't upload, no serial line set, Adafruit Trinket Pro

MrTheSaw, 2018-02-19:
I'm trying to upload to an Adafruit Trinket Pro 5v via the USB port. This works with the Arduino IDE.
In vscode, I can't select a useful serial port. The one the other IDE uses is /dev/cu.usbserial-A6008bj4, but all I see in vscode is the two bluetooth devices.

System: macOS 10.13.3
vscode: 1.20.1
Arduino: 1.8.5
Adafruit AVR Boards: 1.4.12

czgtest, 2018-03-06:
Yes, you can add port in  arduino.json file, like this.

{
    "sketch": "GetStarted.ino",
    "port": "/dev/cu.usbserial-A6008bj4",
    "board": "AZ3166:stm32f4:MXCHIP_AZ3166",
    "configuration": "upload_method=OpenOCDMethod"
}

czgtest, 2018-03-06:
@MrTheSaw  have you resolved this issue ?

MrTheSaw, 2018-03-16:
Well, adding a line to json worked, but is less than awesome.

closed 2018-03-16 by czgtest