microsoft/vscode-arduino #471: Unable select board configuration when the folder in Onedrive

vphuoc, 2018-02-23:
There is no issue if the arduino file outside Onedrive folder. But when I sync it to Onedrive by copping it to Onedrive folder. Then I cannot select the board configuration and can not compile it.

Note that I enabled the File on demand setting in Onedrive app. There is also no issue if the arduino path contains space.

czgtest, 2018-03-02:
@vphuoc Thanks, this issue will be fix next.

czgtest, 2018-03-07:
this is repro step:
1.copy a project to OneDrive
2.use vscode to open the project
3.select the board type on the bottom of toolbar
verify cannot select the board configuration

Sneezry, 2018-03-28:
@vphuoc OneDrive had removed Arduino config file for you to save space. You can require OneDrive always to keep your Arduino file on your machine. See https://github.com/Microsoft/vscode/issues/35433#issuecomment-337382621

czgtest, 2018-03-30:
resolved the problem by turning off Files On-Demand OneDrive feature.  http://www.intowindows.com/what-is-onedrive-files-on-demand-feature/

closed 2018-03-30 by czgtest