microsoft/vscode-arduino #472: Newly added board cannot be displayed

leisiji, 2018-02-24:
I hava installed ESP32 libraries and Arduino IDE can display the board, but vscode cannot display ESP32. However, ESP32 is not installed via Arduino IDE. 
Here is the [installation guide](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md).
[image]
[image]

czgtest, 2018-02-27:
ESP32 is the third party board, we will need to add one URL into the Additional Boards Manager URLs option in vscode User settings.

here is the step:
1.open VSCODE
2.click menu : file -->Preferences-->Settings
3.add code as bellow.  add the URL you want.
 "arduino.additionalUrls": [        
        "https://adafruit.github.io/arduino-board-index/package_adafruit_index.json",
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
    ],

[image]

czgtest, 2018-03-07:
@leisiji  have you got this issue solved?

leisiji, 2018-03-07:
Sorry, I still have not solved this issue. ESP8266 can be found in the list of vscode. However, I cannot find the JSON url in [arduino-esp32](https://github.com/espressif/arduino-esp32). It seems that ESP32 uses the local JSON file.

Sneezry, 2018-03-30:
Hi @leisiji Could you please verify if it works with adding `https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json` into `arduino.additionalUrls` setting?

To add `arduino.additionalUrls` setting, click File -> Preference -> Settings, in USER SETTINGS, add code below:

```
"arduino.additionalUrls": ["https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json"]
```

Then open board manager (press F1 in vscode, and input `Arduino: Board Manager`, then execute the command), and you should be able to find esp32:

[image]

leisiji, 2018-03-31:
Hi, @Sneezry . I have followed your instruction and pressed the button .install, but it showed the error below:
```
Using proxy DIRECT
java.lang.NullPointerException
	at java.util.Collections.sort(Collections.java:175)
	at processing.app.Base.<init>(Base.java:306)
	at processing.app.Base.main(Base.java:140)
[Error] Exit with code=255
```
Moreover, after adding this json, esp32 still cannot be found in Arduino board configuration.

Sneezry, 2018-03-31:
Is the install button on Arduino Extension Board Manager or Arduino IDE?

The error is thrown by Arduino IDE, but not Arduino Extension. Generally it should work by trying twice.

Sneezry, 2018-03-31:
Also, it will help very much if you upload some screenshots.

leisiji, 2018-03-31:
@Sneezry .
The install button is on vscode, Arduino Extension Board Manager.
I have also added the json of esp32 in arduino IDE, 
[image]
but it seems that esp32 cannot be found in board manager of arduino IDE:
[image]
Here is all the error given by vscode ouput window when I pressed the install button in vscode:
```
[Starting] Install package - esp32...
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
正在加载配置...
正在初始化包...
正在准备开发板...
正在下载平台索引... 
Using proxy DIRECT
正在下载平台索引... 已经下载 201kb中的 0kb。
正在下载平台索引... 已经下载 201kb中的 10kb。
正在下载平台索引... 已经下载 201kb中的 20kb。
正在下载平台索引... 已经下载 201kb中的 26kb。
正在下载平台索引... 已经下载 201kb中的 30kb。
正在下载平台索引... 已经下载 201kb中的 40kb。
正在下载平台索引... 已经下载 201kb中的 51kb。
正在下载平台索引... 已经下载 201kb中的 54kb。
正在下载平台索引... 已经下载 201kb中的 64kb。
正在下载平台索引... 已经下载 201kb中的 75kb。
正在下载平台索引... 已经下载 201kb中的 80kb。
正在下载平台索引... 已经下载 201kb中的 90kb。
正在下载平台索引... 已经下载 201kb中的 93kb。
正在下载平台索引... 已经下载 201kb中的 103kb。
正在下载平台索引... 已经下载 201kb中的 106kb。
正在下载平台索引... 已经下载 201kb中的 116kb。
正在下载平台索引... 已经下载 201kb中的 125kb。
正在下载平台索引... 已经下载 201kb中的 135kb。
正在下载平台索引... 已经下载 201kb中的 140kb。
正在下载平台索引... 已经下载 201kb中的 146kb。
正在下载平台索引... 已经下载 201kb中的 156kb。
正在下载平台索引... 已经下载 201kb中的 162kb。
正在下载平台索引... 已经下载 201kb中的 172kb。
正在下载平台索引... 已经下载 201kb中的 178kb。
正在下载平台索引... 已经下载 201kb中的 188kb。
正在下载平台索引... 已经下载 201kb中的 191kb。
正在下载平台索引... 已经下载 201kb中的 201kb。
正在下载平台索引... 
Using proxy DIRECT
正在下载平台索引... 已经下载 0kb中的 0kb。
正在下载平台索引... 已经下载 0kb中的 0kb。
正在下载平台索引... 
Using proxy DIRECT
正在下载平台索引... 已经下载 43kb中的 0kb。
正在下载平台索引... 已经下载 43kb中的 7kb。
正在下载平台索引... 已经下载 43kb中的 13kb。
正在下载平台索引... 已经下载 43kb中的 23kb。
正在下载平台索引... 已经下载 43kb中的 27kb。
正在下载平台索引... 已经下载 43kb中的 33kb。
正在下载平台索引... 已经下载 43kb中的 40kb。
正在下载平台索引... 已经下载 43kb中的 43kb。
正在下载平台索引... 
Using proxy DIRECT
Using proxy DIRECT
正在下载平台索引... 已经下载 6kb中的 0kb。
正在下载平台索引... 已经下载 6kb中的 1kb。
正在下载平台索引... 已经下载 6kb中的 4kb。
正在下载平台索引... 已经下载 6kb中的 6kb。
正在下载平台索引... 
Using proxy DIRECT
java.lang.NullPointerException
	at java.util.Collections.sort(Collections.java:175)
	at processing.app.Base.<init>(Base.java:306)
	at processing.app.Base.main(Base.java:140)
[Error] Exit with code=255
```
I have followed official esp32 installation instruction, and I can change the board in arduino IDE:
[image]

Sneezry, 2018-04-01:
It seems that board cannot install by simply adding an additional URL, here's the official guide: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

czgtest, 2018-08-20:
@leisiji  have your resolve the issue ?

leisiji, 2018-08-20:
@czgtest I tried it on another computer, this issue has gone. But the issue is still on my original computer.

leisiji, 2018-08-22:
Thanks @Sneezry ,  I have solved this problem by installing esp32-arduino in the board manager.

closed 2018-08-22 by leisiji