microsoft/vscode-arduino #474: Enrich device develop experience

chyuanmsft, 2018-02-26:
There might be some improvements on following aspects:
Provide better IntelliSense and Library management experience
Provide well-structured documents and sample code
Provide more project templates for different purpose
Improve experience for developing big/complex project

czgtest, 2018-03-02:
we will consider improving the extension.Thank you for your valuable suggestions.

czgtest, 2018-03-06:
should sync with Chaoyi Yuan about more detail for improvements

elektronikworkshop, 2020-02-05:
Hi, I'm currently implementing the auto IntelliSense feature, see my comment on #438 and others.

robotdad, 2020-10-30:
This looks like a list of things from the previous project team. I think many of these are tracked by other issues, including the one @elektronikworkshop mentions. Please reactivate if I have misunderstood.

closed 2020-10-30 by robotdad