microsoft/vscode-arduino #476: Choose the location of the .vscode folder

paul-arg, 2018-02-26:
I have a main folder that contains multiple folders containing Arduino projects. When I open the main folder with VSCode and then try to upload a sketch to the board, the .vscode folder containing board info and port info is automatically created at the top of the main folder which I don't want because I would like to have a separate .vscode folder for each arduino project subfolder.

Is it possible to choose that the .vscode folder is created in the same folder that contains the .ino main sketch? And not at the top of the workspace?

Thanks

czgtest, 2018-03-02:
If want to have a separate .vscode folder for each arduino project subfolder, we can open the subfolder with vscode and pload a sketch to the board.

Kulbhushan-Chand, 2018-04-01:
@czgtest As you have mentioned each folder can have separate vscode folder. Opening only a single folder in the work-space solves the issue. (Thanks for your suggestion)

However the downside of this approach are - 

**1.** Every-time you update the include path in c_cpp_properties.json file (to get rid of green squiggle lines) in the vscode folder, only that particular file (c_cpp_properties.json) is updated which is in the current workspace folder. (I want to have a common c_cpp_properties.json file for all of my projects so that I do not have to add the include path for cpp intellisence for each of the project folder).
**2.** If you have opened more than one folder in the workspace, each containing its own vscode. Then during verification and uploading the setting are taken from the vscode folder of the topmost folder in the workspace, instead of the current folder whose ino file you have opened.

I hope the asked feature request may come shortly in future.

KolyaKorruptis, 2019-03-26:
I'm also interested in this, for another use case. Not multi-workspace but multi workplace, aka home office:
At my workplace I create multiple tasks for a workspace. According to the recommendation of MS and our own company policy I don't check them into version control. When working remote these tasks are not available. 

If I could define a location for the .vscode folder (say in the workspace settings) I could share the .vscode folder in OneDrive or whatever.