microsoft/vscode-arduino #477: How to keep temporary build folder?

lovettchris, 2018-02-28:
I'm getting this error, but it doesn't tell me how much it overflowed, and it deletes the temporary build folder so I can't even see what it had...  
```
C:\Users\clovett\AppData\Local\Temp\arduino_build_539281/VoiceRecorder.ino.elf section `.text' will not fit in region `FLASH'
```
Can you provide a way to pass all options to Arduino-build ?  For example, I'd like to be able to see verbose compiler output, like I can with the Arudino IDE.

ArthurMa1978, 2018-03-01:
For the build output folder, it can be specified in the project setting
[image]

ArthurMa1978, 2018-03-01:
For the second one, there are 1MB on-board FLASH, if exceeded the ld would print out the number:
[image]

ArthurMa1978, 2018-03-01:
For the last one, currently the Extension not fully support it, we planned to do a big refactor of this extension and provide more like these configurations.

VashJuan, 2018-03-04:
I THINK this should produce verbose compiler output:

Options

The following Visual Studio Code settings are available for the Arduino extension. These can be set in global user preferences Ctrl + , or workspace settings (.vscode/settings.json). The later overrides the former.

{
    "arduino.logLevel": "info", 
}

arduino.logLevel - CLI output log level. Could be info or verbose. The default value is "info".

czgtest, 2018-03-06:
@lovettchris  Have you tried VashJuan 's solution?   have you resolved the issue?

closed 2018-04-17 by czgtest