microsoft/vscode-arduino #478: Build is much slower than Arduino IDE

lovettchris, 2018-02-28:
The Arduino IDE seems to be doing some "incremental" building of core libraries so that when I edit my sketch the "re-verify" happens in seconds.  But with VS code Arduino plugin every verify seems to do the full build and takes minutes.  I would love to get faster incremental build times...

ArthurMa1978, 2018-03-01:
once you specified the build path, it would be "incremental"

v-fional, 2018-03-02:
You can in the project setting to set the build output path.
[image]

TomasHubelbauer, 2018-03-04:
Can this be improved to generate a deterministic implicit build path in temp?

czgtest, 2018-03-06:
Issue #354 has the same question,  I think we need to investigate how to improve Build Manager

aster94, 2018-03-07:
I have also opened an issue #461, suggesting how this should work,  but it has closed without response

sschicken, 2019-10-01:
So I've figured out why this is happening to me. If you use a NFS share on a linux machine, and you are building on a windows machine, the timestamps won't line up properly so the entire cache will be rebuilt every time you verify. To fix this, set your build directory to something local. It's annoying, usually other software that runs into this ends up using Fat File Times to decrease precision a touch and things will line up again, but until/unless that gets implemented the only solution I can see is to change to a local build directory

schlaegerz, 2020-07-02:
I am having this same problem on Windows, even if I have it point the build to a specific folder. Without any code changes it still talks about 1 minute or so to start uploading.

blackpancake, 2021-08-20:
I am having this same problem on Windows, even if I set the "output" path, verifying is still much slower than Arduino IDE.

sahlex, 2022-09-09:
I'm using the arduino-cli for the build task and configured it to build with tasks.json. There you can provide a ```--build-cache-path```. Of course this doesn't work is you just press the 'Verify' button. You have to run the task via the 'Terminal' menu.

It would be helpful, if we could configure that in arduino.json as well...

serfreeman1337, 2023-04-22:
> It would be helpful, if we could configure that in arduino.json as well...

Here is workaround to pass `--build-cache-path` using shell script:
```shell
#!/bin/sh
if [ $1 == "compile" ]; then
    /path/to/arduino-cli $@ --build-cache-path /tmp/arduino_cache_vscode
else
    /path/to/arduino-cli $@
fi
```
Because [arduino-cli is hardcoded](https://github.com/microsoft/vscode-arduino/blob/9544a5feb30a62b7d762306668716606cccf7c7d/src/common/sys/linux.ts#L24), shell script must be also named `arduino-cli`.

Then in extension settings:
- **Arduino: Command Path** -> `arduino-cli`
- **Arduino: Path** -> `/path/to/shell-script`