microsoft/vscode-arduino #481: Command Palette Should Allow (Re)Setting Sketch File

VashJuan, 2018-03-04:
Enhancement request:  Add a new entry to the Arduino Command palette that allows verifying/setting a new/different sketch file. 

It is fairly easy to workaround this by directly editing the .vscode/arduino.json file's 
"sketch": "myMostExcellentSketch.ino"
line, but for newbies this is not immediately obvious.

TomasHubelbauer, 2018-03-04:
This would be very nice, because just today I was trying to find a way to keep multiple indepedent sketches in one repository and ultimately a ended up using multiple repositories due to this. Workspaces do not seem to be supported either, which would have also solved this for me.

czgtest, 2018-03-05:
this is a nice feature, we will sync up with each other and make a plan for it.

czgtest, 2018-03-06:
#382 has the same request,  so we can merge the two issues to fix .

czgtest, 2018-05-14:
User now can click “sketch file” tool bar to input the path of sketch file. 
 
[image]

Notes : the “sketch file” tool bar can show when Arduino.json has set “sketch” value. 
But it is no command palette to allow re-setting sketch file.   Maybe we need to improve it next.

Jackbennett, 2019-11-04:
FYI Other users I had to close and open VScode after installing the plugin to see this specific tooltip in the status bar.

Any chance that dropdown can autosuggest the currently open tabs and fuzzy search .ino,.cpp and .c in suggestions?