microsoft/vscode-arduino #483: Arduino Plugin Moves Terminal Connect Button

tsalinger, 2018-03-05:
--This issue has been moved from [Microsoft/vscode](https://github.com/Microsoft/vscode/issues/44976)--
Author is @MZachmann.
### Issue Type
Bug

### Description

Enable/Initialize the Arduino Extension for this workspace.

Attempt to click the Serial Connect Button on the bottom status bar.

When you go to click it the button saying Win32 appears and the connect button moves to the left.

Click it to connect.

Click the mouse in the terminal window (important!) and the Win32 button will disappear, moving the disconnect button to the right.

Click the disconnect button and instead the Win32 button will appear and you will have to click again.

Unbelievably annoying.

### VS Code Info

VS Code version: Code 1.20.1 (f88bbf9137d24d36d968ea6b2911786bfe103002, 2018-02-13T15:34:36.336Z)
OS version: Windows_NT x64 10.0.16299

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz (8 x 4008)|
|Memory (System)|15.94GB (9.72GB free)|
|Process Argv|C:\Program Files\Microsoft VS Code\Code.exe C:\Users\Mark\OneDrive\Documents\Lora\Pytest|
|Screen Reader|no|
|VM|0%|

</details><details><summary>Extensions (8)</summary>

Extension|Author (truncated)|Version
---|---|---
githistory|don|0.4.0
python|don|0.7.0
gitlens|eam|8.0.2
python|ms-|2018.1.0
cpptools|ms-|0.15.0
debugger-for-edge|msj|0.4.1
vscode-arduino|vsc|0.2.10
markdown-all-in-one|yzh|1.0.5

</details>
Reproduces only with extensions

czgtest, 2018-03-20:
[image]

Sneezry, 2018-03-20:
This issue is caused by using same priority when create serial port status bar item as config status of C/C++ extension.

Higher value of priority means the item should be shown more to the left.

Both Arduino extension and C/C++ extension uses 2 as priority for serial port and config.

Arduino extension:

https://github.com/Microsoft/vscode-arduino/blob/master/src/serialmonitor/serialMonitor.ts#L65

https://github.com/Microsoft/vscode-arduino/blob/master/src/common/constants.ts#L32

C/C++ extension:

https://github.com/Microsoft/vscode-cpptools/blob/aa0f613dfbc32d7d280fbcd033ed5a9b32f078c5/Extension/src/LanguageServer/ui.ts#L32

czgtest, 2018-03-29:
using the latest extension, here is the result now.
1.open vscode and open a existing project
2. Press F1 and select Arduino:Initialize
3. Click the Serial Connect Button on the bottom status bar, and select a port you want.
Verify the Serial Connect Button always in the left and the Win32 button always in the right,  **Fix work.**
4.Click the mouse in the terminal window.
Verify the Win32 button will disappear, "Not Fix"
5.Click the Serial Connect Button again
Verify the Win32 button will appear, the Serial Connect Button need to click again.   "Not Fix"

Sneezry, 2018-03-29:
Win32 button is rendered by C/C++ extension, it is out of our control.

So the 4th and 5th issue won't be fixed.

closed 2018-04-03 by czgtest