microsoft/vscode-arduino #495: "Detected board" incorrect

kyeburchard, 2018-03-10:
Every time I plug my ESP32 Dev Module in with vscode open, I get a suggestion to switch to a different board - it tells me it detected an ESP8266. I have used an 8266 in the past, but the current workspace is configured for the ESP32.
[image]

czgtest, 2018-03-12:
@kyeburchard we will try to investigate it.

czgtest, 2018-03-15:
I plug Huzzah Esp8266 , Sparkfun Esp8266 Thing Dev and Mxchip AZ3166 in with vscode open, 
it can detect the right board and port,  Because I don't have   ESP32 Dev Module board , now I can't repro on the issue.

[image]

[image]

kyeburchard, 2018-03-15:
For me, it correctly identifies my SparkFun ESP8266 Thing Dev, but incorrectly identifies my ESP32 Dev Module as an Adafruit Feather HUZZAH ESP8266. I have tried reinstalling the extension and a few other things. Are there any logs or something I can find and attach?

Sneezry, 2018-05-07:
@kyeb Sorry for response delay. Board detection depends on product ID and vendor ID of your board, so I need know your board's PID and VID. 

You can find them from device manager. Or if you have node installed, you can find them with a simple node script:

```node
const serialport = require('serialport-lite').SerialPortLite;
serialport.list().then(console.log);
```

Install `serialport-lite` by running `npm install serialport-lite`

kyeb, 2018-05-07:
@Sneezry No worries. The response I get from running that node script is the following:
`[ { port: 'COM3', vendorId: 4292, productId: null } ]`
Let me know if I can provide any other info.

Sneezry, 2018-05-08:
It seems your board has the same identity with `Adafruit HUZZAH ESP8266`.

Do you want the extension to popup both `Adafruit HUZZAH ESP8266` and ESP32 when it finds the specific identity, or just remember your choice to stop popup `Adafruit HUZZAH ESP8266` notification?

kyeb, 2018-05-08:
Interesting, I wonder why they would have the same identity...?

Just remembering my choice of "no" and stopping the notification for next time would be awesome (it's really annoying to be notified every time when plugging in and unplugging a bunch).

Sneezry, 2018-05-08:
In fact, PID and VID is not binding to board, but USB control chip. Both HUZZAH and ESP32 use the same chip.

kyeb, 2018-05-08:
I see. The language that "Detected board Adafruit HUZZAH ESP8266" is somewhat misleading in that case.

Sneezry, 2018-05-14:
We have added a new option to ignore specific board when popup board detection in v0.2.15

closed 2018-05-14 by Sneezry