microsoft/vscode-arduino #496: "Arduino Examples" showed twice

reyang, 2018-03-12:
Plug in the DevKit, unplug it, then plug in again, there will be two "Arduino Examples" tabs opened.
Expect only one.

[image]

czgtest, 2018-03-13:
@reyang  which OS are you using ?

I try on my win10 , but not repro this issue
Arduino extension version: 0.2.11
vscode : 1.21.0

reyang, 2018-03-13:
Microsoft Windows [Version 10.0.16299.248]

[image]

Arduino for Visual Studio Code v0.2.11
Azure Account v0.3.1

czgtest, 2018-03-15:
I can repro this issue too.

@Sneezry  Can you help to fix this issue ?

[image]

Sneezry, 2018-03-20:
This issue happens when no document open in vscode. Arduino Examples page is open in the second column (https://github.com/Microsoft/vscode-arduino/blob/0bb2b7392e67b2f59a67382a210fbbd2095a1206/src/extension.ts#L113). When on document open, vscode will open it in the first column, and when we re-plugin DevKit, vscode open a new one in the second column.

Sneezry, 2018-03-20:
I have post an issue here: https://github.com/Microsoft/vscode/issues/46161

ArthurMa1978, 2018-04-09:
Since this is not a issue from our extension, and VS Code planned to  fix it.
Close it on our side.

closed 2018-04-09 by ArthurMa1978