microsoft/vscode-arduino #497: Unable to add '\r\n' when sending text to serial monitor

lirenhe, 2018-03-16:


czgtest, 2018-03-30:
we can select "Both NL & CR" or "No Line ending" now

[image]

closed 2018-04-03 by Sneezry

vishnumaiea, 2021-06-13:
[image]

The line ending character option appear to be missing from my VS Code status bar when then serial monitor is active. Am I missing something? Couldn't find anything related to this at the configurations.

MarcoRBosco, 2021-07-08:
> [image]
> 
> The line ending character option appears to be missing from my VS Code status bar when the serial monitor is active. Am I missing something? Couldn't find anything related to this at the configurations.

Same issue:

[image]

System: Ubuntu 20.04.2 LTS
VS Code: 1.57.1
Arduino plugin: 0.4.3