microsoft/vscode-arduino #500: identifier "GPIOB" is undefined

yzbas123, 2018-03-21:
I don't use any third lib and all code can be found in the local file,I also configure the "includepath"in "c_cpp_properties.json" and all the file can be found by vs code .but it still tell me "identifier 'GPIOB' is undefined",the file with this define is included successfully,please help me.

**INCLUDE PATH CONFIGURATION**
"name": "Win32",
            "includePath": [
                "D:/Visual Studio2017/VC/Tools/MSVC/14.12.25827/include/*",
                "D:/Visual Studio2017/VC/Tools/MSVC/14.12.25827/atlmfc/include/*",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/um",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/ucrt",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/shared",
                "C:/Program Files (x86)/Windows Kits/10/Include/10.0.16299.0/winrt",
                "${workspaceRoot}",
                "${workspaceRoot}/drivers/gpio",
                "${workspaceRoot}/drivers/adp",
                "${workspaceRoot}/drivers/dma",
                "${workspaceRoot}/drivers/aes",
                "${workspaceRoot}/drivers/cmac",
                "${workspaceRoot}/drivers/config",
                "${workspaceRoot}/drivers/eap",
                "${workspaceRoot}/drivers/eax",
                "${workspaceRoot}/drivers/flashif",
                "${workspaceRoot}/drivers/gptimer",
                "${workspaceRoot}/drivers/i2c",
                "${workspaceRoot}/drivers/mmu",
                "${workspaceRoot}/drivers/network",
                "${workspaceRoot}/drivers/mac",
                "${workspaceRoot}/drivers/phy",
                "${workspaceRoot}/drivers/plc",
                "${workspaceRoot}/drivers/uart",
                "${workspaceRoot}/drivers/spi",
                "${workspaceRoot}/drivers/sysctrl",
                "${workspaceRoot}/drivers/systick",
                "${workspaceRoot}/drivers/wdt",
                "${workspaceRoot}/drivers/rtc",
                "${workspaceRoot}/startup",
                "${workspaceRoot}/include",
                "${workspaceRoot}/utillity",
                "${workspaceRoot}/mdk"
            ]
**PROBLEM**

message: 'identifier "GPIOB" is undefined'
at: '120,10'
source: ''
code: 'undefined'

Sneezry, 2018-03-21:
Hi @yzbas123 , have you tried include Arduino built in header files?

As C/C++ vscode extension doesn't support ecursive path in `includePath` current, we can use `browse` as workaround.

Could you try `c_cpp_properties.json` with content below?

```
{
  "configurations": [
    {
      "name": "Win32",
      "browse": {
        "path": [
          "${env:programfiles(x86)}\\Arduino\\hardware",
          "${env:programfiles(x86)}\\Arduino\\libraries",
          "${env:userprofile}\\Documents\\Arduino\\hardware",
          "${env:userprofile}\\Documents\\Arduino\\libraries",
          "${workspaceRoot}"
        ]
      },
      "intelliSenseMode": "clang-x64"
    }
  ],
  "version": 3
}
```

Please notice that the config above is not official suggestion. Official suggestion is to use `includePath` (as you do), but `includePath` is not ecursive currently, we use `browse` as a workaround here.

Sneezry, 2018-03-28:
@yzbas123 is the issue resolved? If yes, I will close this issue.

closed 2018-03-29 by Sneezry