microsoft/vscode-arduino #505: Errouse Arduino VS code extension in a proxy environment

ihere1, 2018-03-28:
Reported by @kwaxi on azure-iot-developer-kit gitter

Error message:
```
[Starting] Update package index files...
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
Preparing boards...
Downloading platforms index... 
Using proxy DIRECT
java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
    at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
    at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
    at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
    at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
    at processing.app.Base.<init>(Base.java:298)
    at processing.app.Base.main(Base.java:144)
Caused by: java.net.ConnectException: Connection refused: connect
    at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
    at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:85)
    at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
    at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
    at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
    at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
    at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
    at java.net.Socket.connect(Socket.java:589)
    at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
    at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
    at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
    at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
    at sun.net.www.http.HttpClient.New(HttpClient.java:339)
    at sun.net.www.http.HttpClient.New(HttpClient.java:357)
    at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
    at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
    at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
    at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
    at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
    at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
    at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
    ... 5 more
[Error] Exit with code=255
```

Sneezry, 2018-03-28:
> `Error downloading http://downloads.arduino.cc/packages/package_index.json`

I believe this issue is caused by connection to downloads.arduino.cc on user's side.

ArthurMa1978, 2018-04-04:
Got same issue reported before, after remove everything under 
C:\Users\[user name]\AppData\Local\Arduino15
problem solved.

ArthurMa1978, 2018-04-04:
From  @kwaxi : 
I've just found out that the issue might have been sole a proxy issue. Arduino IDE comes with a bundled java runtime, and this didn't contain the ca certificates of the root from our ssl-offloading proxy

closed 2018-04-04 by ArthurMa1978

ctastan, 2018-07-09:
just change proxy setting to "No proxy" via Arduino IDE.
File->preferences->Network->No proxy
Resolved for me.