microsoft/vscode-arduino #512: Any Arduino command results in 'b.forEach is not a function'

basovink, 2018-04-02:
help wanted

Yesterday I was progamming away on my mac with VSCode and vscode-arduino. Today any Arduino command (ex ept set serial port and serial monitor) results in 'b.forEach is not a function'.

I don't recall changing anything. Also I have by now uninstalled VSCode and removed all .vscode directories I could find.

Totally lost here.

czgtest, 2018-04-03:
@basovink , Can you repro this issue now after re-install vscode and vscode-arduino ?  can you give more detail about what you do?

basovink, 2018-04-03:
It is totally reproducible. 

A did a fresh install, removing any trace before with 'AppCleaner'. The installed a freshly downloaded VSCode and installed the extensions: openHab and vscode-arduino. Nothing Arduino from the command palette works, except 'set serial port' and 'serial monitor'.

The console tells me:
```
TypeError: b.forEach is not a function
	at Object.union (/Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/common/util.js:294:7)
	at pkg.platforms.forEach (/Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/arduino/boardManager.js:181:49)
	at Array.forEach (native)
	at rawModel.packages.forEach (/Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/arduino/boardManager.js:175:27)
	at Array.forEach (native)
	at BoardManager.loadPackageContent (/Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/arduino/boardManager.js:174:27)
	at BoardManager.<anonymous> (/Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/arduino/boardManager.js:54:22)
	at Generator.next (<anonymous>)
	at /Users/bas/.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.11/out/src/arduino/boardManager.js:9:71
	at Promise (<anonymous>)
e.$onExtensionRuntimeError	@	mainThreadExtensionService.ts:44
e._doInvokeHandler	@	rpcProtocol.ts:174
e._invokeHandler	@	rpcProtocol.ts:159
e._receiveRequest	@	rpcProtocol.ts:108
e._receiveOneMessage	@	rpcProtocol.ts:90
(anonymous)	@	rpcProtocol.ts:33
(anonymous)	@	rpcProtocol.ts:213
e.fire	@	event.ts:142
(anonymous)	@	ipc.net.ts:81
emitOne	@	events.js:96
emit	@	events.js:191
readableAddChunk	@	_stream_readable.js:178
Readable.push	@	_stream_readable.js:136
onread	@	net.js:560
```

Sneezry, 2018-04-03:
@basovink Could you try to clear your Arduino package folder and try again? The default path on macOS is `~/Library/Arduino15`. Additional URL will be remove as well, so backup it before the action.

basovink, 2018-04-03:
Thanks @Sneezry! This was the right solution.

closed 2018-04-03 by basovink