microsoft/vscode-arduino #524: Can't build with 0.2.12 and vscode 1.2.22

MZachmann, 2018-04-13:
Projects that used to build fine don't build any more. I found that if I moved a project to the root folder it would build. Otherwise I get "file not found" messages if I build in my standard folder (which is much farther down the tree).

If I had to guess, something in a path is getting messed up because it's too long and so my longer paths are failing.

typical message: "readlink c:\Users\Mark\OneDrive\Documents\MyArduinoFolder: The system cannot find the file specified."

but the folder does exist and this build worked yesterday and if I move the entire folder to the root it does build

MZachmann, 2018-04-15:
I've tried to reproduce this and it's entirely flaky. My test is manually running Arduino-builder in the command line and changing only the path to the sketch.

I can build in the folder C:\ff but not in the folder C:\FeatherNrf52. In the second folder the SketchLoader fails as it recursively enumerates the folders with a file in use error. The two folders have identical contents.

Here's a dump of the command line output.

C:\t>"C:\Program Files (x86)\Arduino\arduino-builder" -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware C:\Users\Mark\AppData\Local\Arduino15\packages -hardware "C:\Users\Mark\Documents\Sketch\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools C:\Users\Mark\AppData\Local\Arduino15\packages -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Mark\Documents\Sketch\libraries" -fqbn=adafruit:nrf52:feather52:softdevice=s132v201,debug=l0 -ide-version=10805 -build-path c:\Users\Mark\Documents\f5build -warnings=none -prefs=build.path=c:\Users\Mark\Documents\f5build -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\Mark\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.nrfjprog.path=C:\Users\Mark\AppData\Local\Arduino15\packages\adafruit\tools\nrfjprog\9.4.0 -verbose c:\Feather\FeatherNrf53.ino -debug-level 10
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
===info ||| Using board '{0}' from platform in folder: {1} ||| [feather52 C%3A%5CUsers%5CMark%5CAppData%5CLocal%5CArduino15%5Cpackages%5Cadafruit%5Chardware%5Cnrf52%5C0.8.3]
===info ||| Using core '{0}' from platform in folder: {1} ||| [nRF5 C%3A%5CUsers%5CMark%5CAppData%5CLocal%5CArduino15%5Cpackages%5Cadafruit%5Chardware%5Cnrf52%5C0.8.3]
===warn ||| WARNING: Category '{0}' in library {1} is not valid. Setting to '{2}' ||| [Wireless+Communications LightLora Uncategorized]
readlink c:\Feather: The system cannot find the file specified.

C:\t>"C:\Program Files (x86)\Arduino\arduino-builder" -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware C:\Users\Mark\AppData\Local\Arduino15\packages -hardware "C:\Users\Mark\Documents\Sketch\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools C:\Users\Mark\AppData\Local\Arduino15\packages -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Mark\Documents\Sketch\libraries" -fqbn=adafruit:nrf52:feather52:softdevice=s132v201,debug=l0 -ide-version=10805 -build-path c:\Users\Mark\Documents\f5build -warnings=none -prefs=build.path=c:\Users\Mark\Documents\f5build -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\Mark\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.nrfjprog.path=C:\Users\Mark\AppData\Local\Arduino15\packages\adafruit\tools\nrfjprog\9.4.0 -verbose c:\ff\FeatherNrf53.ino -debug-level 10 | more
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
===info ||| Using board '{0}' from platform in folder: {1} ||| [feather52 C%3A%5CUsers%5CMark%5CAppData%5CLocal%5CArduino15%5Cpackages%5Cadafruit%5Chardware%5Cnrf52%5C0.8.3]
===info ||| Using core '{0}' from platform in folder: {1} ||| [nRF5 C%3A%5CUsers%5CMark%5CAppData%5CLocal%5CArduino15%5Cpackages%5Cadafruit%5Chardware%5Cnrf52%5C0.8.3]
===warn ||| WARNING: Category '{0}' in library {1} is not valid. Setting to '{2}' ||| [Wireless+Communications LightLora Uncategorized]
===info ||| Progress {0} ||| [66.67]
_id=feather52
bootloader.tool=bootburn
build.arch=NRF52
build.board=NRF52_FEATHER
... more valid output here

Sneezry, 2018-04-16:
Hi @MZachmann Do you save your Arduino project in OneDrive folder? From the log, it seems you save your Arduino project at `c:\Users\Mark\OneDrive\Documents\MyArduinoFolder`, right?

If you'd like to sync your Arduino project with OneDrive, it should be better to ask OneDrive always keep these files on you local machine.

More details can be found here: https://github.com/Microsoft/vscode/issues/35433#issuecomment-337382621

MZachmann, 2018-04-16:
I am sensitive to the fact that OneDrive does strange things with regards to locking files. For the second post (real testing this stuff) I paused OneDrive and (as you can see in the post) moved folders out of the OneDrive folder.

In this issue I put the source folder into C:\ff and C:\Feather so not in OneDrive. 

However... I think you can close this issue if you want. As far as I can tell this is a bug in Arduino-builder not in anything in the VScode or the Arduino shell. You may have made a change the makes the bug more obvious but i'm fairly sure it's not your bug.