microsoft/vscode-arduino #529: avrdude cannot open "libreadline.so.6"

Myl0g, 2018-04-16:
Whenever I try to upload to my Arduino Uno, I get this error message:

```
/home/milo/.arduino15/packages/arduino/tools/avrdude/6.0.1-arduino2/bin/avrdude: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory
```
Everything *was* working until this message popped up out of the blue. I've already tried reinstalling the extension and the actual Arduino software. I'm running Arch Linux.

Thanks in advance for any advice you may have!

Myl0g, 2018-04-16:
Also: the actual Arduino IDE has no trouble uploading.

czgtest, 2018-04-17:
@Myl0g , can you try the solution in bellow article? 

http://forum.arduino.cc/index.php?topic=481449.0

Myl0g, 2018-04-18:
And of course the issue mysteriously resolves itself.

Thanks for your help anyway!

closed 2018-04-18 by Myl0g

Myl0g, 2018-04-18:
Update: the issue was fixed by updating the Arduino AVR board from the Arduino IDE's board manager. The reason the fix didn't get reflected in vscode was because the config files in the .vscode workspace folder weren't up-to-date. Deleting them and re-initializing the project is what fixes the issue in vscode.