microsoft/vscode-arduino #530: TestingOpen output on the serial monitor

f6fvy, 2018-04-16:
What's the point of sending "TestingOpen" in the Serial monitor window after opening it ?

It look me a while to understand it was coming from this plugin (after digging the sources), and not from my Arduino code !

Sneezry, 2018-04-17:
TestingOpen is for test if serial port open successfully.

f6fvy, 2018-04-17:
I understand, but maybe it would be suitable to make this text sending optional (and/or allow its modification by the user).

Sneezry, 2018-04-17:
That makes sense. I will add a new setting to allow user to disable it.

czgtest, 2018-05-11:
this fix work on version 0.2.15

closed 2018-05-11 by czgtest

Sneezry, 2018-05-14:
We have added a new config (`arduino.disableTestingOpen`) to disable auto testing open in v0.2.15, see https://github.com/Microsoft/vscode-arduino#options