microsoft/vscode-arduino #533: Arduino: Examples command displays an empty tab

Kurolox, 2018-04-17:
Basically, whenever I try to use the Arduino: Examples command it displays an empty tab [like this.](https://i.imgur.com/fb6Do9X.png)

The arduino path in the settings is set up as ```    "arduino.path": "/usr/share/arduino"```, and the examples are in ```/usr/share/arduino/examples```. the Arduino IDE lists the examples correctly.

VSCode version and information:
```
Version 1.22.2
Commit 3aeede733d9a3098f7b4bdc1f66b63b0f48c1ef9
Date 2018-04-16T09:43:25.785Z
Shell 1.7.12
Renderer 58.0.3029.110
Node 7.9.0
Architecture x64
```

Arduino plugin version: 0.2.12

Sneezry, 2018-04-18:
What is your operating system? Is it Ubuntu?

Kurolox, 2018-04-18:
Arch Linux. Here's the output of ```uname -a```.

```Linux Horus 4.15.15-1-ARCH #1 SMP PREEMPT Sat Mar 31 23:59:25 UTC 2018 x86_64 GNU/Linux```

jhenninger, 2018-04-30:
The examples work when pointing `arduino.path` to an Arduino IDE downloaded from the official website.

When the Arduino IDE installed by Arch's package manager is used (`/usr/share/arduino`), the extensions tries to access `/usr/share/arduino/libraries` and throws an exception because that directory does not exist. 

If you create an empty `libraries` directory in /usr/share/arduino`, vscode correctly shows the built-in examples.

jhenninger, 2018-05-03:
Hmm, your folder looks exactly like mine. 

Can you take a loot at the developer console (Help -> Toggle Developer Tools) to see what error you are getting when opening the Arduino examples tab?

Here is what I get when the libraries directory does NOT exist: https://i.imgur.com/Nqij6B0.png

Kurolox, 2018-05-04:
Well, this is bizarre. It seems like it's looking for ~/Arduino/libraries for some reason: https://i.imgur.com/lGKHX8W.png

I have explicitly configured vscode to use the /usr/share/arduino path with `    "arduino.path": "/usr/share/arduino"`, but for some reason it's looking in there. I don't even have an `Arduino` folder in home.

Any idea about what could be happening?

jhenninger, 2018-05-05:
Afaik the offical Arduino IDE creates that folder.

Kurolox, 2018-05-06:
That still doesn't explain why it's looking for the folder in that path when I've explicitly set it up otherwise on the vscode settings, though. It's still looking at the default path for some reason. Could you share your vscode config (at least the config regarding this plugin) to check if there's any difference?

czgtest, 2018-06-06:
this issue is fix on 0.2.16, please update the latest version and use it. Thanks a lot for your feedback

closed 2018-06-06 by czgtest