microsoft/vscode-arduino #535: Need Help for "Upload using Programmer"

amien-moh, 2018-04-20:
I have litlle project with MigtyCore variant.

for overall feature don't have any problem.

I have issue with "upload using programmer", 
but I don't know it's supported by  extension.

ubuntu 17.10 x64
Arduino 1.8.5
extension Arduino 0.2.12

Thank's

czgtest, 2018-04-25:
this issue is dulplicate with #407 , so close it

czgtest, 2018-05-10:
@amien-moh  vscode-arduino-0.2.15-rc is support to upload using programmer, can you have a try ？

steps:
1.download from here
https://github.com/Microsoft/vscode-arduino/releases/download/v0.2.15-rc/vscode-arduino-0.2.15-rc.vsix 
2.open vscode and install vscode-arduino-0.2.15-rc from VSIX
3.Press F1 and select arduino: upload using programmer

amien-moh, 2018-05-13:
Thanks very much...
get and test mode right now... :+1:

closed 2018-05-30 by czgtest