microsoft/vscode-arduino #536: arduino basic installation

MHz000, 2018-04-22:
Issue Type: <b>Bug</b>

Starting VS code I got the message:
can not find the arduino installation path
As an complete newbie don't know what to do

Extension version: 0.2.12
VS Code version: Code 1.22.2 (3aeede733d9a3098f7b4bdc1f66b63b0f48c1ef9, 2018-04-12T16:38:45.278Z)
OS version: Windows_NT x64 10.0.16299

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-7260U CPU @ 2.20GHz (4 x 2208)|
|Memory (System)|7.89GB (3.93GB free)|
|Process Argv|C:\Program Files\Microsoft VS Code\Code.exe|
|Screen Reader|no|
|VM|0%|

</details>

czgtest, 2018-04-25:
@MHz000 ，Arduino IDE is required. Please install it from here.  
https://www.arduino.cc/en/main/software#download
Please try again after install Arduino IDE.

MHz000, 2018-04-25:
Hi Chen,
arduino IDE is already installed and works fine with esp8266. VScode didn't
find esp-specific .h files like #include<esp8266WiFixxx> nor are esp-boards
in board select menue shown. Any idea what is wron, what is missing in my
setup? c_ccp_properties.json looks like:
           {
            "name": "Win32",
            "includePath": [
                "C:/Dropbox/ARDUINO Sketch book/libraries",
                "C:/Benutzer/Win10/AppData/Local/Arduino15",
                "${workspaceFolder}"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "intelliSenseMode": "msvc-x64",
            "browse": {
                "path": [
                    "C:/Dropbox/ARDUINO Sketch book/libraries",
                    "C:/Benutzer/Win10/AppData/Local/Arduino15",

"C:/Benutzer/Win10/Dokumente/ArduinoData/packages/esp8266/tools",

"C:/Benutzer/Win10/Dokumente/ArduinoData/packages/arduino/tools",

"C:/Benutzer/win10/Dokumente/ArduinoData/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/scr",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": ""
            },
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
Thank-you!

czgtest, 2018-04-25:
1.open vscode 
2. add arduino url in user settings , like this.

{
     "arduino.additionalUrls": [
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json",
        "https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json"
    ]
}

3. press F1 and select arduino: board Manager
4. install esp8266 package

MHz000, 2018-04-25:
> package_azureboard_index.json"
>    1. press F1 and select arduino: board Manager
>    2. install esp8266 package

*BINGO! Chen you got it!!!*
following your excellent and detailed instruction I see now all the ESP
boards! Thank you so much.
However one small step still missing in my setup. My source code contains
the following includes:
#include <ESP8266WiFiMulti.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
VSc tells me "Include file not found in browser path" Edit browser path
eddings"
My browser phath looks like:
{
"name": "Win32",
"includePath": [
"C:/Dropbox/ARDUINO Sketch book/libraries",
"C:/Benutzer/Win10/AppData/Local/Arduino15",
"${workspaceFolder}",
"C:\\Users\\Win10\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\
esp8266\\2.0.0\\cores\\esp8266"
],
"defines": [
"_DEBUG",
"UNICODE",
"_UNICODE"
],
"intelliSenseMode": "msvc-x64",
"browse": {
"path": [
"C:/Dropbox/ARDUINO Sketch book/libraries",
"C:/Benutzer/Win10/AppData/Local/Arduino15",
"C:/Benutzer/Win10/Dokumente/ArduinoData/packages/esp8266/tools",
"C:/Benutzer/Win10/Dokumente/ArduinoData/packages/arduino/tools",
"C:/Benutzer/win10/Dokumente/ArduinoData/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/scr"
,
"${workspaceFolder}",
"C:\\Users\\Win10\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\
esp8266\\2.0.0\\cores\\esp8266"
],
"limitSymbolsToIncludedHeaders": false,
"databaseFilename": ""
},

Pls notice my arduino sketch book is located at: "C:/Dropbox/ARDUINO Sketch
book",
inside is the library.  "C:/Dropbox/ARDUINO Sketch book/libraries",

*May you have a look at this settings and  give advice for correct
settings.*
*:-)) thank-you Chen for beeing patient with a newbie*

closed 2018-04-27 by czgtest