microsoft/vscode-arduino #540: Any command can't be found

r3d9u11, 2018-05-02:
Hi. I have VSCode 1.22.2 and installed extension "vscode-arduino" via extension manager. But any command "Arduino: *" will be crashed. For example: "Arduino: Initialize" will rise exception "command 'arduino.initialize' not found". What I'm doing wrong?

czgtest, 2018-05-03:
Arduino IDE is required. Please install it from here.
https://www.arduino.cc/en/main/software#download

r3d9u11, 2018-05-03:
@czgtest yes, it's installed (unpacked from tar.xz, I'm on Linux Mint), latest version. also setting.json and arduino.json configured, too. (but latest version of arduino ide haven't subdirectory bin and scriptbrun-arduini.sh)

r3d9u11, 2018-05-03:
[development](https://github.com/Microsoft/vscode-arduino#development) version works fine

closed 2018-05-03 by r3d9u11