microsoft/vscode-arduino #547: autodefine paths for intellisence

r3d9u11, 2018-05-04:
Hi. Is it possible automatically detect and fill "generic" paths like a:
```
...
"<...PATH_TO_ARDUINO...>/hardware/tools/avr/avr/include",
"<...PATH_TO_ARDUINO...>/hardware/tools/avr/lib/gcc/avr/4.9.2/include",
"<...PATH_TO_ARDUINO...>/hardware/arduino/avr/cores/arduino",
"<...PATH_TO_ARDUINO...>/hardware/arduino/avr/variants/mega",
...
```
in c_cpp_properties.json after type of board was selected? It would be cool. Thanks!

Sneezry, 2018-05-09:
This is a very very old issue, but it is still not solved completely: https://github.com/Microsoft/vscode-arduino/issues/438

In short, `includePath` is not recursive currently, and it depends on C/CPP extension (which is out of our control).

I had commited a workaround to fix it temporary: https://github.com/Microsoft/vscode-arduino/pull/538

This issue is very annoying, and we will keep to seek final solution.

r3d9u11, 2018-05-10:
@Sneezry thank you for your answer and useful links! (sorry for duplicate, I can close this issue. or leave it opened for other users who interested in this theme)

Sneezry, 2018-05-14:
Then I will close this issue. Please post any update in #438 to follow this issue.

closed 2018-05-14 by Sneezry