microsoft/vscode-arduino #562: command 'arduino.upload' not found

scott-linenberger, 2018-05-17:
Receive an 'arduino.<command name>' not found when attempting to do anything (upload, verify, use boared manager) with the extension. Arduino path is set,  Arduino 1.8.5 is installed and working.

muyuuuu, 2020-02-06:
where the `.vscode` folder

ZEROIOT, 2020-02-11:
Thanks, I ran into the same problem and solved it by your method now
1. I installed the arduino plugin and added the arduino IDE path. I created a new arduino project and found that no matter it was compiled or uploaded: command 'arduino.xxx' not found. At the same time, I found that in the lower right corner of my vs code software interface, some serial port selection and board selection options did not appear. Just like the software interface before the plug-in is installed.
2. According to your method:
I closed the vs code software and found the installation location of the plugin: C: \ Users \ user \ .vscode \ extensions
Deleted the vsciot-vscode.vscode-arduino-0.2.29 folder (this is the arduino plugin installed)
3. Open the vs code software again, open a folder containing the .ino project, and open the .ino file. At this time, the vs code software intelligently detects the file type. It is recommended to install the arduino plug-in. After clicking OK to install, all problems are resolved.
The above are my steps, I hope that can help people with the same problem, thanks again