microsoft/vscode-arduino #564: Board definition on Atmega 2560 memory reporting issue

MrBambix, 2018-05-21:
Hi,

Seems the board definition for an 2560 AVR board is not right, cannot upload a 72k sketch allthough 256k is possible...

regards,
Leo

MrBambix, 2018-05-21:
Sketch uses 74776 bytes (231%) of program storage space. Maximum is 32256 bytes.
Global variables use 2880 bytes (140%) of dynamic memory, leaving -832 bytes for local variables. Maximum is 2048 bytes.
===error ||| Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it. ||| []

MrBambix, 2018-05-21:
Board is correctly set to 2560 btw

czgtest, 2018-05-31:
@MrBambix  Can you try to upload your sketch on Arduino IDE ?  is it work ?

MrBambix, 2018-05-31:
Hi,

Yes it is not a code problem it is definitely the board definitions reporting the available memory on an atmega 2560 board as too low. Sketch works fine on the Arduino IDE, compiles, uploads and Arduino IDE is reporting available memory on the 2560 just fine.

 

Thanks for looking into this.

 

Regards

Leo

czgtest, 2018-06-06:
@MrBambix ,  the board definition for an 2560 AVR board is in board.txt file of package folder. 
For Windows, the default path is C:\Users\{UserName}\AppData\Local\Arduino15\packages\emoro\hardware\avr\3.2.2\boards.txt.

Both Arduino IDE and VSCode Arduino Extension will use the same board defination when upload.

So please check if you have select the right board in Arduino Extension and use the same board definition with Arduino IDE.

[image]

czgtest, 2018-06-12:
@MrBambix  any update from you ?

closed 2018-08-17 by czgtest