microsoft/vscode-arduino #571: on verify cmd, first 30 lines of open file will be destroyed

apollo2mond, 2018-05-26:
Issue Type: <b>Bug</b>

unclear ...
seems that there is an issue with temporary buf/cache, if i change something than the change is also not considered with verify cmd and after vsc restarting the file is still partly destroyed.
in the current setup it can reproduce the scenario but it worked before for days without this behaviour

Extension version: 0.2.15
VS Code version: Code 1.23.1 (d0182c3417d225529c6d5ad24b7572815d0de9ac, 2018-05-10T17:11:17.614Z)
OS version: Windows_NT x64 6.3.9600

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz (4 x 2594)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: disabled_software<br>video_decode: enabled<br>video_encode: enabled<br>vpx_decode: unavailable_software<br>webgl: enabled<br>webgl2: enabled|
|Memory (System)|15.70GB (1.44GB free)|
|Process Argv|E:\uController\VS Code\Code.exe|
|Screen Reader|no|
|VM|0%|

</details>

czgtest, 2018-08-17:
@apollo2mond  I can't repro on this issue,  I add 2 line code and not saved before verify cmd,  

then I run  arduino: verify cmd , new add 2 line code will be saved and verify command run successfully,
not see rubbish in the first line of the file.

Can you give more information if you still can repro on the issue?

[image]

[image]

czgtest, 2018-09-04:
Closing due to inactivity. If you still see this problem, please reactivate.

closed 2018-09-04 by czgtest