microsoft/vscode-arduino #572: Where I found the .hex file?

juloko, 2018-05-26:


juloko, 2018-05-26:
I already found at #367. This shoud works at first:

In the arduino.json file put:
`"output": ".\\bin"`

But I have to remove all that I made, removing the folder ".vscode". Settings and run the command:
`"Arduino: Initialize"`

closed 2018-05-26 by juloko

AnielShri, 2018-12-05:
@juloko : Thanks! Was also looking for the output .hex files; your second comment was quite useful

Zoldyck99, 2021-09-21:
> I already found at #367. This should works at first:
> 
> In the arduino.json file put:
> `"output": ".\\bin"`
> 
> But first, I had to remove all that I made, removing the folder ".vscode". Settings and run the command:
> `"Arduino: Initialize"`

This worked for me too, much thanks.