microsoft/vscode-arduino #577: i verify and revived this error :Exit with code=undefined

Alisparrow2000, 2018-06-04:
help help
don't know what to do ????

Sneezry, 2018-06-04:
Did you install Arduino IDE? If yes, did you install it from Windows App Store?

Alisparrow2000, 2018-06-05:
yes i installed arduino IDE .no i installed from arduino;s web site

Sneezry, 2018-06-05:
This error usually appears when the extension cannot find Arduino IDE. The extension looks for Arduino IDE in two places:

1. `arduino.path` in user settings of VSCode
2. default path of Arduino IDE. For Windows, the default path is `C:\Program Files (x86)\Arduino`.

So please check if you have set an incorrect Arduino path in VSCode user settings (open VSCode, **File** - **Preferences** - **Settings**). If you didn't set `arduino.path` in VSCode user settings, check if Arduino IDE is installed in `C:\Program Files (x86)\Arduino`.

Alisparrow2000, 2018-06-05:
i checked every thing you said but the issue didnot solve

eknowlton, 2018-06-22:
I get this issue as well, but on mac. I may be configuring it wrong.

I've managed to get it working on Mac with:

```
{
    "arduino.path": "/Applications/Arduino.app",
    "arduino.commandPath": "Contents/MacOS/Arduino",
}
```

motycakjan, 2018-06-24:
Hi, this setting works for me, where the _C_Cpp.intelliSenseEngine_ was the main reason of my issue. 

`{
    "arduino.path": "/Applications",
    "arduino.commandPath": "Contents/MacOS/Arduino",
    "arduino.additionalUrls": "",
    "arduino.logLevel": "info",
    "C_Cpp.intelliSenseEngine": "Tag Parser"
}`

fengpakho, 2018-06-25:
I don't understand.

fengpakho, 2018-06-25:
Don't send this mail to me

motycakjan, 2018-06-28:
I did not send you this email, this is notification email from this channel.

> You are receiving this because you are subscribed to this thread.

eknowlton, 2018-06-28:
Lmao

czgtest, 2018-08-16:
@Alisparrow2000 , the error was caused by command path,   please remove "arduino.commandPath": "run-arduino.bat" in your vscode settings.  restart vscode after change.

arduino.commandPath" is  only option setting, the default value is "C:/Program Files (x86)/Arduino/arduino_debug.exe" on windows ,  so we can not set it .
if you set  "arduino.commandPath": "run-arduino.bat", you should put "run-arduino.bat" file in path "/Program Files (x86)/Arduino" , then run arduino command,  the error will resolve.

closed 2018-08-16 by czgtest

Desertgeek, 2018-10-14:
I ran into the same problem in my Arduino UNO project that I am developing on my MacBook Pro. I can confirm that the suggestion offered by eknowlton above works and allows the package index to update successfully. I also incorporated the suggestion by motycakjan re: the intellisense engine.

human2010, 2019-04-26:
@Alisparrow2000 Is your problem solved? I also encountered the same problem, please help!

Josh-BI-UK, 2020-04-06:
> [image]
> help help
> don't know what to do ????
==========================================
I had the same issue using windows 10 and resolved it as follows:

1) In Visual Studio Code, click **File > Preference > Settings**.

2) Type “Arduino” in the search textbox; the Arduino settings page should display.

3) In the field entitled [**Arduino: Command Path**] simply type in "**_arduino_debug.exe_**", nothing more.

I found that if I typed in the full path, VSCode would prefix the base file path again, automatically, ending up with an incorrect path like this;

C:\Program Files (x86)\Arduino\C:\Program Files (x86)\Arduino\arduino_debug.exe

Simply using "**_arduino_debug.exe_**", resolved the problem for me.

[image]

Akshay1595, 2020-12-14:
> I had the same issue using windows 10 and resolved it as follows:
>
> 1. In Visual Studio Code, click **File > Preference > Settings**.
> 2. Type “Arduino” in the search textbox; the Arduino settings [...]

This worked for me as well! THanks