microsoft/vscode-arduino #578: Use programmers from board definitions

maxekman, 2018-06-05:
I noticed that the Hologram.io Dash board uses a custom programmer in the Arduino IDE (see screenshot). I'm not familiar with how the Arduino IDE handles this, but it seems like the additional programmers in comes with the board definition. Currently the programmers in the VSCode extension are hard coded to the default programmers in Arduino.

Is there any plan for supporting programmers from the board definition?

[image]

czgtest, 2018-06-06:
@maxekman , I don't have enough knowledge about your custom programmer, I suggest you can git clone code of vscode-arduino and add your code to support it, we will review your pull request. maybe it is a solution.

maxekman, 2018-06-06:
Thanks for the advice! I’ll see what I can do.

hairlesshobo, 2019-02-01:
I am curious where VSCode pulls its list of programmers from. I too am working with a custom programmer. It is defined in the programmers.txt file in C:\Program Files (x86)\Arduino\hardware\arduino\avr, and is picked up just fine by the Arduino IDE. I also have a couple custom board definitions in boards.txt in the same directory, and VSCode sees those no problem. 

I only added the programmers AFTER installing the Arduino VSCode extension. A quick assumption would be that VSCode keeps the list cached and doesn't refresh this list once it has been initially populated. I had hoped that restarting VSCode, or reinstalling the Arduino extension would allow them to appear, but they did not.

In the past using a VisualStudio extension for Arduino, there was a menu entry that allowed you to refresh the Arduino toolchain so that new board and programmer definitions could be recognized. Either a manual refresh, or at least to have these definitions refreshed on VSCode launch, like the Arduino IDE does, would seem necessary.

Out of curiosity, is there a way to clear VSCode's Arduino cache? If so, maybe I could try doing this and it would allow my programmer to appear.

Thanks in advance!

hairlesshobo, 2019-02-02:
I have done exactly that and modified the extension to automatically read programmers from the arduino platforms. I have submitted the pull request, but this is my first time contributing to an existing github project, so I hope I did the pull request process correctly.

closed 2019-05-29 by stale[bot]