microsoft/vscode-arduino #579: Adding Arduino Core STM32 Boards for Debug

ChriD, 2018-06-05:
Hi,

I have a problem adding a board to debug.
I added
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
as additional board manager.
 I can compile and upload sketches. But i can not debug the NUCLEO-F303RE eval board with the message

> "Debug for board Nucleo-64 is not supported by now."

I have added those lines to usbmapping.json
It seems Nucleo-64 subboards do have all the same pid?
How do i get the correct pid and vis. I used the values from this link
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L78

```

 {    
    "index_file": 
    "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json",
    "boards": [
      {
        "vid": "0483",
        "pid": "5711",
        "name": "Nucleo_64",
        "package": "STM32",
        "architecture": "STM32F3",
        "id": "NUCLEO_F303RE",
        "target": "stm32f3x.cfg"
      }
      ]
    },
```

and this to debuggerUsbMapping.json

```
 {
        "vid": "0483",
        "pid": ["374b","5711"],
        "name": "ST Link V2.1",
        "short_name": "stlink-2.1",
        "config_file": "stlink-v2-1.cfg"
    },

```

my arduino.json looks like this
```
{
    "board": "STM32:stm32:Nucleo_64",
    "configuration": "pnum=NUCLEO_F303RE,upload_method=MassStorage,xserial=third,usb=none,opt=osstd",
    "sketch": "Blink.ino"
}

```

How can i get this board debugged?
What am i missing?

czgtest, 2018-06-06:
@ChriD  which OS do you use ?

ChriD, 2018-06-06:
Hi @czgtest 
I am using Windows 10 with newest VSCode and newest vscode-arduino plugin

czgtest, 2018-06-21:
@ChriD You can use this tools https://github.com/EmergingTechnologyAdvisors/node-serialport to list the vid/pid:
npm install -g serialport
serialport-list -f jsonline

ChriD, 2018-06-21:
Thanks! I'll try!

closed 2018-08-16 by czgtest