microsoft/vscode-arduino #582: Upload Using a Status Bar Button

momiccioli, 2018-06-08:
How do I get a button on the Status Bar for "Arduino Upload"?  I use this all the time as I would expect most do.  This should be there by default.

czgtest, 2018-06-11:
@momiccioli now you can upload using command "Arduino:upload" in the Command Palette (F1 or Ctrl + Shift + P) ,  we not add button in status bar for uploading.

momiccioli, 2018-06-11:
I realize I can do that but still want to add to the status bar.  I use the upload command more than any other when developing.  Its too bad to have to go through multiple actions for something that should be able to do in a single click.

sstaub, 2018-06-20:
I think there should be two buttons, one for upload and one for verify.

czgtest, 2019-01-21:
you can see verify and upload button in action bar when select a .ino file

[image]

closed 2019-01-21 by czgtest