microsoft/vscode-arduino #596: COM port issue

lamqhoang, 2018-06-20:
I have issues with the COM port recently. 
The COM port list does not show up when I click on the "Select Serial Port". I can set the COM port in the setting json and upload to Arduino board without issue. But if I try to open the port, the error message below show up.

> Failed to open serial port COM4 due to error: + TypeError: Cannot read property 'close' of null

[image]
VScode and arduino extension are updated to latest version. The COM port is not in used

czgtest, 2018-06-21:
@lamqhoang  VSCode insider will repro on this issue, Can you use VSCode stable ? 

VSCode insider has been upgraded to node.js >8.0 , This will have breaking changes on the native module on vscode arduino. we will fix this issue next.

czgtest, 2018-07-06:
The Nodejs of VSCode insider is reverted to 7.9, so it work well now.

closed 2018-07-06 by czgtest

kjcolley7, 2018-12-20:
I believe this issue is present again in recent VSCode Insiders versions. Happens to me on 1.31.0-insider for Mac.

Snippet from Developer Tools:

```
TypeError: Cannot read property 'close' of null
	at Object.<anonymous> (bindings.js:27)
	at Object.<anonymous> (bindings.js:46)
	at Module.t._initNodeRequire.r._compile (loader.js:721)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:722)
...
```

This exception is generated from the node-usb-native library.

Sneezry, 2018-12-20:
@kjcolley7 this issue has been fixed with https://github.com/Microsoft/vscode-arduino/pull/731, and you can use Arduino extension in insiders with next release.

bluefate, 2019-06-23:
Getting the error in latest insiders version.
Failed to open serial port COM3 due to error: + TypeError: Cannot read property 'close' of null

doddydigitaldesign, 2019-06-30:
I'm having the same error as @bluefate .
VSCode-Insiders version: 1.36.0

q10359443, 2019-07-05:
[image]

q10359443, 2019-07-05:
I found the same mistake
VSCode-Insiders version: 1.36.0-insider
[image]

VSCode  version: 1.36.0
[image]

q10359443, 2019-07-05:
@lamqhoang @czgtest

j3rrey, 2019-07-05:
Newest VSCode and Arduino plugin.
I'm not able to choose the correct port.
In the Arduino Ide everything works as expected
[image]

viniciusczanini, 2019-07-05:
Have anyone found the solution for this issue?

rido-min, 2019-07-05:
rollback VSCode to 1.35

viniciusczanini, 2019-07-05:
Thank you very much, rollback to 1.35 did the trick!

Here's the link if anyone need:
https://code.visualstudio.com/updates/v1_35

danieldiasm, 2019-07-07:
Yeah, 1.35 User worked out for me too. Thanks!!

DanwithArduino, 2019-07-08:
1.36 generates no response when trying to change serial port, but 1.35 and 1.34 crashes the arduino extension when trying to change the port.

huba71, 2019-11-11:
Unfortunately 1.40 breaks the com port again. 1.39.2 seems to work. (on Mac OSX at least)

gcurtis79, 2020-03-22:
Still getting this with VSCode Windows and Linux:

VSCode: 1.43.1
Arduino: 0.2.29

It worked fine a couple weeks ago...

Mechanikerlegion511, 2020-12-01:
ВОТ зачем делать обновления после которых не чего не работает?????????????????