microsoft/vscode-arduino #598: Serial Monitor as a standalone extension

stigbjorlykke, 2018-06-24:
It should be possible to install/use the Serial Monitor as a standalone extension and make vscode-arduino depend on this. The Serial Monitor is useful for more than Arduino development and the <Select Programmer> and <Select Board Type> in the Status Bar is not applicable in this environments.

fengpakho, 2018-06-26:
？

czgtest, 2018-06-28:
It's a great piece of advice,  Thank you for your feedback.

peter73421, 2018-11-16:
any progress on this

danpeig, 2019-08-11:
Agree!

lewisxhe, 2020-01-04:
Agree! +1

robotdad, 2020-10-30:
Interesting suggestion, I agree it is useful in other scenarios.

gcampbell-msft, 2023-02-16:
@stigbjorlykke We have taken your advice and the serial monitor is now in it's own extension and Arduino depends on it! Please check out the most recent version of the Arduino extension to try it out.

closed 2023-02-16 by gcampbell-msft