microsoft/vscode-arduino #610: `arduino.path` doesn't seem to expand variables

kankaristo, 2018-07-17:
None of these settings are working for me (I'm restarting VSCode between these):
```json
"arduino.path": "$home/.bin/arduino",
"arduino.path": "$HOME/.bin/arduino",
"arduino.path": "/home/$user/.bin/arduino",
"arduino.path": "/home/$USER/.bin/arduino",
```

But if I give a fixed path, it works fine:
```json
"arduino.path": "/home/sami/.bin/arduino",
```

I'm running Linux (Fedora).

BTW, does the extension find `arduino` if it's in `PATH` (I don't currently have it there)?

czgtest, 2018-07-20:
@kankaristo , Yes, before use this extension,  we need to install Arduino IDE,  and set  the arduino path in Visual Studio Code settings.  then it will work.

kankaristo, 2018-07-25:
Ah, I thought this was a bug (`$user` is used in the readme as well), but I guess it's a feature request instead. :stuck_out_tongue:

Is this an "upcoming" feature, or why does the readme use `$user`? Or is it meant to be understood as "insert your username here" (in which case using `$` is a bit confusing).

And BTW, I'm not sure how VSCode packages typically expand variables, and if they actually use *environment* variables, or some other type of variable expansion. Because `$user` isn't an environment variable (`$USER` is, though), but some other VSCode packages do use `$home` (instead of `$HOME`) which seems to be working fine.

I share my config across a few different computers (a [dotfiles](https://dotfiles.github.io/) repository), but I don't use the same username on all of them. This is the only section of my VSCode config that contains my username (I have Arduino in my home folder, I guess I could move it to `/opt/arduino/` or similar).