microsoft/vscode-arduino #611: " [Error] Exit with code=undefined " -> too little information

gajop, 2018-07-19:
I'm getting the error as mentioned (`[Error] Exit with code=undefined`), which is lacking information.
My `arduino.path`, `arduino.commandPath` or something else is probably wrong, but it's too hard to tell where the issue is.
Can you please increase the verbosity of the commands like Verify, so we have a chance to figure out what went wrong?
Thanks!

czgtest, 2018-07-20:
@gajop  Thank you for your feedback. 

before use this extension,  we should install Arduino IDE. 
you can not set  “arduino.path”  , when you use this extension ,  it will  automatically detected from your Arduino IDE installation path. 

arduino.commandPath is not required too , it is only option.

for this error, we will improve the description,  Can you tell us the detail of how to repro on it ?

gajop, 2018-07-20:
I don't have Arduino IDE installed in the default path, which is why I specified a custom `arduino.path` pointing to the right location.

I think you can easily reproduce this by specifying any bogus `arduino.path`. This is not a bug, just lacks verbosity to help the user figure out what settings are wrong.

czgtest, 2018-08-15:
if not install Arduino IDE or set incorrect path for  `arduino.path` ,  then you run arduino command , will get an message as bellow.

[image]

if `arduino.commandPath` is not right,  we will get the error " [Error] Exit with code=undefined "  when run arduino command.  Thank you a lot for your feedback,  we will improve the error information then

czgtest, 2018-08-17:
[image]

closed 2018-08-17 by czgtest

gajop, 2018-08-17:
I'd add some space after the ".", but otherwise it's a good error message (especially as it also includes the attempted path)

human2010, 2019-04-26:
@czgtest  " [Error] Exit with code=undefined " ，I also encountered such a problem, how to solve it? Please give pointers!

PeterWone, 2019-09-21:
I don't _have_ a commandPath specified but I'm getting code=undefined, what else can it mean?