microsoft/vscode-arduino #613: read content of keywords.txt for text highlights

aster94, 2018-07-21:
hello,
arduino uses a file inside the library folder caller "keywords.txt" used to generate some highlights for the function's names, costants, class and more

please see:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#keywords

here there is the arduino core keywords.txt:
https://github.com/arduino/Arduino/blob/master/build/shared/lib/keywords.txt

here an example from a library i wrote:
https://github.com/aster94/GoProControl/blob/master/GoProControl/keywords.txt

VincentChalnot, 2018-08-12:
+1 This would be a very interesting feature :+1:

aster94, 2018-08-14:
What do you mean exactly with "dynamical"? Would it be possible to load it only at startup of vscode? 

Arduino IDE load all the keywords.txt files only at startup, a similar behaviour would be ok

Sneezry, 2018-08-14:
Keywords needs be pre-defined in the extension in package.json, but not load from external files.

See https://code.visualstudio.com/docs/extensionAPI/language-support#_syntax-highlighting

aster94, 2018-08-20:
mmm clear, ok let's see what can be done

elektronikworkshop, 2020-02-19:
Just saw this as I'm currently working hard on a proper IntelliSense implementation (currently alpha testing), see #438 and [my branch notes](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md). This will become obsolete as soon as latter is integrated. `keywords.txt` is a dirty hack of the Arduino IDE to work around the lack of something powerful as IntelliSense.

Added it to my list of related issues.

Regards,  
EW

robotdad, 2020-10-30:
Agree with @elektronikworkshop, IntelliSense features, which need improvement, should provide the same but better functionality than the keywords.txt file. Reactivate if there is something I'm missing in this request.

closed 2020-10-30 by robotdad