microsoft/vscode-arduino #614: highlights errors

aster94, 2018-07-21:
in the arduino ide there is a very useful soft red highlight of the first error encountered during compile time:

[image]

you should add this and overtake it highlighting all the errors.
the second error in this example is not highlighted as you can see

[image]

czgtest, 2018-07-23:
It is a great feature request. Thank you for your feedback

aster94, 2018-07-23:
Thanks

I was also thinking that maybe highlighting only the first error would be faster since the compiler stops there instead of checking all the code. Maybe you could add an option to leave the user choose if he wants all the code checked or having it stopping at the first one

aster94, 2018-09-03:
you may find this useful:
https://github.com/Microsoft/vscode-extension-samples/tree/master/decorator-sample

closed 2019-05-29 by stale[bot]

smarthomeagentur, 2020-03-16:
Is there any way to highlight errors on compile? I tested it with a fresh project and the only way to find the compilation error seems to be the output in terminal