microsoft/vscode-arduino #616: Set default Baud Rate

The33Coder, 2018-07-23:
Hello,

I read the README but I don't find how to change the default Baud Rate. My default Baud is `115200` and I want to have `9600`. The only solution when I find, is to change this in serialMonitor.js...  (#395)

Is there any other solution?

Sneezry, 2018-07-25:
Done.

czgtest, 2018-08-17:
this feature is enable in latest version 0.2.20, please set the default band rate in vscode user settings.

"arduino.defaultBaudRate": 115200

closed 2018-08-17 by czgtest

etekweb, 2023-08-10:
This issue has re-emerged. All of the code related to this PR was removed in v0.5.0 ([this commit](https://github.com/microsoft/vscode-arduino/commit/bb0bcd43f2e7b43cdc00df369be3dda7cbae7bae) was when it was removed from the documentation). However, the Serial Monitor baud rate is still changing automatically to 115200 every time the Arduino is flashed with the latest version (0.6.0).

Is there a new way to fix this?

yenmangu, 2024-01-16:
> This issue has re-emerged. All of the code related to this PR was removed in v0.5.0 ([this commit](https://github.com/microsoft/vscode-arduino/commit/bb0bcd43f2e7b43cdc00df369be3dda7cbae7bae) was [...]

Why were they removed? I would like to see this being re added in, I would hate to move BACK to platformIO because of this

Ares9323, 2024-04-29:
Workaround: 
in your user folder, open `.vscode\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\out\src\serialmonitor\serialMonitor.js`

and change line 22 to:
`this.lastSelectedBaudRate = 9600; // Same default as Arduino.`

Will probably need to redo this when they'll update the extension but it's better than setting the baud rate after every upload 😅