microsoft/vscode-arduino #618: Out of Workspace Includes

ssnover95, 2018-07-25:
Hello,
I recently started using this extension so it's possible what I'm trying to do is supported and I've made a mistake somewhere, but I can't find an example with this element.

I have a header file which is from another project that I would like to include in my sketch. I don't want to move it into the workspace directory and would rather just specify the include path. I was able to get the IntelliSense to behave and find the file correctly by modifying the `c_cpp_properties.json` file, but when I try to build the executable, it errors and says that it can't find the file. Is there a way to specify extra include paths to the compiler as well?

ssnover95, 2018-07-25:
In `c_cpp_properties.json`, I have as one of the entries in `includePath`: `"/home/ssnover/libraries/RF24/**"`. In my sketch I include `RF24.h` and in the directory that's included there is a file RF24.h. I know there's not a small mistake in that part since the IntelliSense is working (I'm able to control-click and it opens the correct file).

I've attempted to execute that command and then executed the Verify command to build and I receive the same error:
```
[Starting] Verify sketch - nrf-transmitter.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/home/ssnover/develop/home-automation/testing/nrf-transmitter/nrf-transmitter.ino:3:18: fatal error: RF24.h: No such file or directory
 #include "RF24.h"
                  ^
compilation terminated.
exit status 1
[Error] Exit with code=1
```

Is it possible to view the Makefile that's being run or some of the commands being fed to the compiler so I can debug this issue more easily?

pjrp, 2019-02-09:
Any developments on how to solve this?
There must be an easy way of adding project specific libraries.

mMerlin, 2019-07-22:
Currently there does not appear to be a way to do what you want.  The entries in `c_cpp_properties.json` for `includePath` tell intellisense where to look.  They do *NOT* tell the compiler.  There does not seem to be a makefile (as such) saved where I could find it.  However, if you have added an `output` path entry in `.vscode/arduino.json` for the protect, go there and look at `build.options.json`.  The interesting entry is `otherLibrariesFolders`, which points ONLY to the libraries folder of the Arduino IDE specified sketchbook folder.  It looks like specifying multiple folders for that entry would do the trick, but there is no way currently to pass that through from the extension configuration settings.

If you are working on a system (OS and file system) that supports symbolic links, here is a work around that should work (not tested).  Create a folder (outside of your project) to use as a dummy sketchbook folder, with a nested libraries folder.  Create symbolic links in that folder to each of the library folders needed by the project.  Change the Arduino IDE sketchbook.path entry (`$HOME/.arduino15/preferences.txt` on my system) to point at the dummy sketchbook folder.  It should not be too difficult to automate backing up, changing, restoring the preferences.txt file.  Your project folder could be a child of the created sketchbook folder, making the libraries folder a sibling.  Doing it that way, simply pointing the Arduino IDE to that sketchbook folder should let it work directly with your project.  As well as the vscode arduino extension.

ObieOne1, 2020-01-27:
Wauw, so I have just spent 2 days messing around with includepaths and browse.paths before I stumbled on your post. Thanks for this. I will not search any further for now and keep my includes just in the root folder for now.

elektronikworkshop, 2020-02-18:
IntelliSense is not linked to Arduino's build process. If you want Arduino's builder to find code outside your sketch, you have three choices (best choices first):
1. provide a library (this is easier than you might think)
1. blend the code into your workspace using either
   1. symbolic links (when you're on any UNIX system like Linux or OSX)
   1. manage your additional project with a SCM like git and clone into the workspace
1. if it's a header: include it by its relative path
   1. from a sub-directory of your sketch (recommended, but doesn't fit the above requirement)
   1. from a parent directory or by the absolute path (both not recommended)

Arduino's [build interface](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) doesn't allow to set additional search directories except for the path of your current  sketch's and Arduino's sketchbook library path.

**Maintainers, @dooriya:** This issue can be closed as it won't fix. This is a characteristic/limitation of Arduino - every user has to live with or work around it.

dooriya, 2020-02-19:
Thanks @elektronikworkshop for the information.

closed 2020-02-19 by dooriya

elektronikworkshop, 2020-02-19:
You're welcome!

psuchris, 2023-04-03:
> Currently there does not appear to be a way to do what you want. The entries in `c_cpp_properties.json` for `includePath` tell intellisense where to look. They do _NOT_ tell the compiler. There does [...]

In case anyone else ends up here, the process to change your default sketchbook directory is now different with arduino-cli because `otherLibrariesFolders` in `build.options.json` does not come from the 'preferences.txt' file. Instead it comes from a configuration file that is not created by default.

The configuration key that needs updated (see [https://arduino.github.io/arduino-cli/0.31/configuration/](https://arduino.github.io/arduino-cli/0.31/configuration/)) is `directories: user:`.

Steps to generate and edit the configuration file (Windows directories):

1. Navigate to `C:\Users\[username]\.vscode\extensions\vsciot-vscode.vscode-arduino-0.6.0-win32-x64\assets\platform\win32-x64\arduino-cli`
2. Open a terminal in this location. (On Windows, hold SHIFT and right-click, then select "Open PowerShell/Command Prompt window here")
3. Run the command: `./arduino-cli config init`. It will tell you where the file was generated. For me, this was `C:\Users\[username]\AppData\Local\Arduino15\arduino-cli.yaml`
4. In the generated `arduino-cli.yaml` file, edit the `user:` value under `directories:` and save.

Note: There is also a `builtin.libraries:` key that can be specified under `directories:` that can potentially be an additional library directory. I tried a few different formats to specify multiple values for the `user:` key in the .yaml file but they all gave me errors after rebuilding the IntelliSense configuration (CTRL + ALT + i) in VSCode.