microsoft/vscode-arduino #628: OpenOCD doesn't get killed when terminating debugging while target running

DeqingSun, 2018-07-29:
When terminating debugging while target running, OpenOCD won't get killed. So VScode will not debug again unless I do "pkill openocd" in another terminal.

Here us debug console output

Terminating debugging while target running
```
C disconnect: {"restart":false}
```

Terminating debugging while target is halted in a breakpoint
```
C disconnect: {"restart":false}
1: (7840) <-1020kill
1: (7840) ->&"kill\n"
1: (7840) ->~"Kill the program being debugged? (y or n) [answered Y; input not from terminal]\n"
E output: {"category":"stdout","output":"openocd: Info : dropped 'gdb' connection\n","data":null,"type":"output"}
1: (7840) ->=thread-group-exited,id="i1"
openocd: Info : dropped 'gdb' connection
1: (7841) ->1020^done
1: (7841) ->(gdb)
E output: {"category":"stdout","output":"Kill the program being debugged? (y or n) [answered Y; input not from terminal]\n","data":null,"type":"output"}
Kill the program being debugged? (y or n) [answered Y; input not from terminal]
1: (7843) 1020: elapsed time 2
1: (7843) <--gdb-exit
1: (7843) ->^exit
1: (7846) "/Users/sundeqing/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-gdb" exited with code 0 (0x0).
1: (7847) <-logout
1: (7848) <-logout
E output: {"category":"console","output":"The program '/Users/sundeqing/Documents/Github/deqingsmallproject/purePCB/D11Board/Blink_test_no_BL/.build/d11c14a/Blink_test_no_BL.ino.elf' has exited with code 0 (0x00000000).\r\n\n","data":null,"type":"output"}
The program '/Users/sundeqing/Documents/Github/deqingsmallproject/purePCB/D11Board/Blink_test_no_BL/.build/d11c14a/Blink_test_no_BL.ino.elf' has exited with code 0 (0x00000000).

E exited: {"exitCode":0,"type":"exited"}
E terminated: {"type":"terminated"}
E output: {"category":"telemetry","output":"VS/Diagnostics/Debugger/DebugCompleted","data":{"VS.Diagnostics.Debugger.ImplementationName":"Microsoft.MIDebugEngine","VS.Diagnostics.Debugger.EngineVersion":"14.0.50719.1","VS.Diagnostics.Debugger.HostVersion":"14.0.50719.1","VS.Diagnostics.Debugger.AdapterId":"cppdbg","VS.Diagnostics.Debugger.DebugCompleted.BreakCounter":2},"type":"output"}
 R: {"success":true,"message":null,"request_seq":13,"command":"disconnect","body":null,"running":false,"refs":null,"seq":0,"type":"response"}
```

alexwhittemore, 2019-02-28:
Running into this same problem, just to add visibility.

iitaka1142, 2019-10-30:
I got same problem. input "-exec monitor shutdown" then you can shutdown openocd from gdb. unfortunately gdb down, too. you have to restart debug mode manually.