microsoft/vscode-arduino #630: Upload Using Programmer with USBtinyISP without a serial port set.

Seairth, 2018-08-04:
When attempting to program with a USBtinyISP, I am receiving the following error:

`Please specify the upload serial port.`

USBtinyISP does not use a serial port.  In this case, the port should not be checked or passed as a command-line option.

feltte, 2018-12-15:
I have same problem with release 0.2.24.
Is this issue fixed?

CaptainMack, 2019-04-17:
No updates for this? I'm having the same problem.

tobia, 2020-06-07:
So what's the solution?

hellyzh, 2020-06-09:
Setting an arbitrary serial port allows the sketch to be uploaded without appearing to cause any issue. You can select a serial port and rerun.

closed 2020-06-09 by hellyzh

feltte, 2020-06-09:
Why closed?
Some PC have no serial port and USBasp is not recginized as a serial port. I cannot select an arbitrary serial port because there is no serial port.

a7hybnj2, 2020-06-09:
Agree with @feltte I have 1 serial and its already in use by another utility and when I select it it acts strangely.

hellyzh, 2020-06-10:
Add a line `"port": "port"` to the config file ./.vscode/arduino.json. Then it will assume you have selected a port named port. To prompt user to select a port, it is because most of cases, it need to choose a serial port to do operation like open serial port or upload code.