microsoft/vscode-arduino #635: output path : no such files or directory

AntoineGirafe, 2018-08-09:
Hi there, I'm trying to set the ouput path for arduino build but somehow it's not working. I get the error : 

```
avr-g++: error: Board_to_waspmote_LoRaWAN.ino.cpp: No such file or directory
avr-g++: warning: '-x c++' after last input file has no effect
avr-g++: fatal error: no input files
compilation terminated.
```
there is the Json files :

```
{
    "board": "arduino:avr:pro",
    "configuration": "cpu=8MHzatmega328",
    "sketch": "Board_to_waspmote_LoRaWAN.ino",
    "output": "../build",
    "port": "COM3"
}
```

When I remove the "output" line from the Json files it works ...  Any Idea ?

Sneezry, 2018-08-14:
I cannot reproduce this issue

[image]

Does it work if you change `../build` to `./build`? Also try to create the output folder first and try again.

AntoineGirafe, 2018-08-14:
I tried  both it's still not working ... in fact the folder is created but I don't know why it can't reach it... I tried to moove my workspace somewhere else but I get always the same error :-/

czgtest, 2018-08-17:
Can you have a try "output": "./.build" ?

KazW, 2018-09-09:
I'm running into the same issue, but it says one of my headers is missing. If I delete the given output folder, it rebuilds from scratch successfully, but then subsequent builds fail with the "No such file or directory" error. Absolute and relative paths produce the same result.

czgtest, 2018-10-09:
Add some logs to check the output path.

if the output path is not exist,  it will pop up  a message
 
[image]

If the output path is exit, will print out in Arduino channel
[image]

closed 2018-10-09 by czgtest

danindiana, 2018-11-08:
Where do I find the settings.json file in Ubuntu 16.04? 

 OS: Ubuntu 16.04 xenial
 Kernel: x86_64 Linux 4.15.0-38-generic
 Uptime: 3h 39m
 Packages: 3512
 Shell: bash 4.3.48
 Resolution: 2560x1080
 DE: Unity 7.4.5
 WM: Compiz
 WM Theme: Radiance
Radiance [GTK2]
, Radiance [GTK3]
 Icon Theme: ubuntu-mono-light
 Font: Ubuntu 11
 CPU: AMD Ryzen 5 1600X Six-Core @ 4GHz
 GPU: GeForce GTX 1050
 RAM: 9940MiB / 16034MiB