microsoft/vscode-arduino #641: arduino.path instruction not working for Ubuntu

n9iels, 2018-08-19:
The README of this plugin states the following about the `arduino.path`:
> arduino.path - Path to Arduino, you can use a custom version of Arduino by modifying this setting to include the full path. Example: C:\\Program Files\\Arduino for Windows, /Applications for Mac, /home/$user/Downloads/arduino-1.8.1 for Linux. (Requires a restart after change). The default value is automatically detected from your Arduino IDE installation path.

I noticed that these instructions are not working for Ubuntu 18.04. For some reason the `$user` (or `$USER`) variable is not used. The plugin keeped complaining about a invalid `arduino.path` till I changed this variable to my username.

czgtest, 2018-08-20:
@n9iels  Thank you for your feedback,  `arduino.path` doesn't  expand variables now,  we will update readme file later.

czgtest, 2018-08-20:
The arduino.path instruction has changed as bellow.
/home/<username>/Downloads/arduino-1.8.1 for Linux.

closed 2018-08-20 by czgtest

tzaroon, 2019-03-02:
Just install:
sudo apt install python
pip install pyserial

everything works fine because arduino IDE works on python2.xx in ubuntu