microsoft/vscode-arduino #643: #include statements inside if blocks marked as necessary dependecies

435vic, 2018-08-21:
In many libraries, if blocks are used to support older versions of Arduino, like this:
```
#if (ARDUINO >= 100)
 #include "Arduino.h"
#else
 #include "WProgram.h"
#endif
```

If my version of arduino is recent enough, `Arduino.h` will be looked for. Otherwise, It'll be `WProgram.h`.
The issue is, VSCode looks for both, and as WProgram.h was *replaced* with Arduino.h, one of them wil not be found, producing a warning.

I would say to just ignore any `include`s inside if blocks, or ignore the ones that don't get run, such as the `#else` (if my arduino is recent)

relu91, 2018-08-24:
I have the same issue with Encoder.h library.

antti202, 2019-05-30:
Would be nice to get fixed, still an issue.

adiazulay, 2020-12-11:
This was a bug with c/c++ extension and should be fixed now. Please reopen this issue if you're still experiencing the problem.

closed 2020-12-11 by adiazulay