microsoft/vscode-arduino #644: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper

JulianAssmann, 2018-08-22:
Whenever I start VS Code on Ubuntu 18.04 LTS with the Arduino Plug installed, I get the same error multiple times:

`[Starting] Update package index files...
Exception in thread "main" java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.desktop/java.awt.Toolkit.newAWTError(Toolkit.java:472)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:488)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
	at java.base/java.util.HashMap$KeySpliterator.forEachRemaining(HashMap.java:1608)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at java.desktop/java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:532)
	at java.desktop/java.awt.Toolkit.getDefaultToolkit(Toolkit.java:613)
	at java.desktop/java.awt.Toolkit.getEventQueue(Toolkit.java:1494)
	at java.desktop/java.awt.EventQueue.invokeLater(EventQueue.java:1312)
	at java.desktop/javax.swing.SwingUtilities.invokeLater(SwingUtilities.java:1421)
	at arduinopc.main(arduinopc.java:83)
Caused by: java.lang.ClassNotFoundException: org.GNOME.Accessibility.AtkWrapper
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:374)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:485)
	... 15 more
No required groups are missing for this user.
Exception in thread "main" java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
	at java.desktop/java.awt.Toolkit.newAWTError(Toolkit.java:472)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:488)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:177)
	at java.base/java.util.HashMap$KeySpliterator.forEachRemaining(HashMap.java:1608)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at java.desktop/java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:532)
	at java.desktop/java.awt.Toolkit.getDefaultToolkit(Toolkit.java:613)
	at processing.core.PApplet.<clinit>(Unknown Source)
	at processing.app.Preferences.save(Preferences.java:735)
	at processing.app.Preferences.init(Preferences.java:249)
	at processing.app.Base.main(Base.java:117)
Caused by: java.lang.ClassNotFoundException: org.GNOME.Accessibility.AtkWrapper
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:374)
	at java.desktop/java.awt.Toolkit.fallbackToLoadClassForAT(Toolkit.java:485)
	... 15 more
[Done] Updated package index files.`

Additionally I have to input the password for my account as many times as the error occurs with the message

`Authentification is required to add the user to the required groups`

Shmukers, 2018-08-23:
I've had the same issue with Kubuntu 18.04 when launching Umlet this morning. It appears to be caused by an unattended security upgrade for the openjdk packages.

The [answer by rdrever to this askubuntu SE question](https://askubuntu.com/questions/695560/assistive-technology-not-found-error-while-building-aprof-plot) fixed the error.

Quoting the fix:
> This can be done by editing the accessibility.properties file for OpenJDK:
> `sudo vim /etc/java-8-openjdk/accessibility.properties`
> Comment out the following line:
> `assistive_technologies=org.GNOME.Accessibility.AtkWrapper`

Hope this helps.

komlenic, 2018-08-25:
The solution by @Shmukers posted above solves issues related to other software as well.  I found this via Google while searching for help resolving this issue when attempting to run Netbeans on Ubuntu 18.04/Mint 19.

czgtest, 2018-08-27:
@JulianAssmann  do you have a try the solution ?

sdennler, 2018-08-29:
Running on OpenJDK 11 the file is
/etc/java-11-openjdk/accessibility.properties

jaromrax, 2018-09-03:
I just installed arduino in ubuntu 18.04 - I see this error. 
In the file `/etc/java-11-openjdk/accessibility.properties` I commented the only line
`assistive_technologies=org.GNOME.Accessibility.AtkWrapper`
and `arduino` started the ide.

closed 2018-09-04 by czgtest

mellertson, 2019-08-17:
> I've had the same issue with Kubuntu 18.04 when launching Umlet this morning. It appears to be caused by an unattended security upgrade for the openjdk packages.
>
> The [answer by rdrever to this [...]

Also worked for me running Ubuntu 18.10.

Ravi-jetani, 2019-09-04:
> I've had the same issue with Kubuntu 18.04 when launching Umlet this morning. It appears to be caused by an unattended security upgrade for the openjdk packages.
>
> The [answer by rdrever to this [...]

also worked for me in Ubuntu 18.10

aztlan2k, 2020-02-16:
The proposed solution helped me get `groovysh` running on Chrome OS.  Thanks @terryspring !

> This can be done by editing the accessibility.properties file for OpenJDK:
> `sudo vim /etc/java-8-openjdk/accessibility.properties`
> Comment out the following line:
> `assistive_technologies=org.GNOME.Accessibility.AtkWrapper`

Muhammedfasil, 2021-02-23:
> I've had the same issue with Kubuntu 18.04 when launching Umlet this morning. It appears to be caused by an unattended security upgrade for the openjdk packages.
>
> The [answer by rdrever to this [...]

worked for me, thanks

dginovker, 2021-03-01:
Is the accepted solution _not_ working for anybody else? (Kubuntu 20.04)

JayDi85, 2021-03-13:
> Is the accepted solution not working for anybody else? (Kubuntu 20.04)

Try to find another `accessibility.properties` file in the system.

robozb, 2021-07-31:
> I've had the same issue with Kubuntu 18.04 when launching Umlet this morning. It appears to be caused by an unattended security upgrade for the openjdk packages.
>
> The [answer by rdrever to this [...]

Thank you so much, this worked at: 

Linux 3.16.0-4-amd64 #1 SMP Debian 3.16.51-3 (2017-12-13) x86_64 GNU/Linux

With Tomcat 8

CarlosEmanuelMartins, 2021-12-15:
> > This can be done by editing the accessibility.properties file for OpenJDK:
> > `sudo vim /etc/java-8-openjdk/accessibility.properties`
> > Comment out the following line:
> > `assistive_technologies=org.GNOME.Accessibility.AtkWrapper`
> 
> Hope this helps.

This will solve the bug found when tring to install chatty on ubunto 20.04 as of 15 December of 2021.

Zungzwange, 2022-04-30:
I wrote the same code but didn't work i need this Revager :( what should I do help!!

Exception in thread "main" java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
at javax.swing.RepaintManager.<clinit>(RepaintManager.java:238)
at javax.swing.JComponent.repaint(JComponent.java:4792)
at java.awt.Component.repaint(Component.java:3311)
at javax.swing.JLabel.setText(JLabel.java:345)
at javax.swing.JLabel.<init>(JLabel.java:161)
at javax.swing.JLabel.<init>(JLabel.java:194)
at org.revager.tools.GUITools.getMessagePane(GUITools.java:534)
at org.revager.Main.main(Main.java:117)