microsoft/vscode-arduino #645: should IntelliSense suggest only built in function?

aster94, 2018-08-23:
hello,

as wikipedia sais the aim of intelliSense is to

> reducing the amount of name memorization needed

i think that the intelliSense in vscode should be "focused" on the built-in functions used by the arduino users and not in all the functions which are hidden by the core

example:
[image]
i guess that no one ever used that stuff, the common user is interested only in pinMode

[image]
it doesn't even suggest delay

[image]
same for the serial, where is "begin"?

many and many other examples could be made
the point is that the arduino's user uses arduino not to use the function inside the core but only the built-in which can be found in Arduino.h
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Arduino.h

[image]
also the libraries inside the core shouldn't been suggested, the only suggested should be the libraries inside the workspace and the one from the core (https://github.com/arduino/ArduinoCore-avr/tree/master/libraries)

i don't know if this is possible/easy but this would be another great improvements

czgtest, 2018-08-27:
Thanks a lot for your feedback and suggestion,   it is a good feature request to improve intelliSense.

aster94, 2018-10-26:
I don't have any knowledge about intellisense but i think this is funny: the perfect behaviour should be this one
[image]

but unfortunately using the normal way arduino people code it doesn't show anything
[image]

closed 2019-05-29 by stale[bot]