microsoft/vscode-arduino #664: Cannot click Select Serial Port

Nickiel12, 2018-10-01:
whenever I click the Select Serial Port button in the bottom right, I get "Extension Host terminated unexpectedly. with two buttons, Open Dev Tools, and Restart Extension Host. This happens every time, and I haven't found anything online

czgtest, 2018-10-08:
@Nickiel12  Do you use the latest VSCode ? 
Now I use vscode 1.27.2,  the serial port work well.

please see #590

Nickiel12, 2018-10-08:
Yes 1.27.2 User Setup. Are you using the system wide? I can't get that to work in a specific scenario, so I use the user setup.

czgtest, 2019-01-21:
@Nickiel12  Can you have a try on v0.2.25?  we add serial port support for Electron 3.0.

Nickiel12, 2019-01-22:
the com button works perfectly now, thank you!!

closed 2019-04-09 by czgtest

davidamayuelas, 2020-03-13:
It doesn't work for me. I'm just trying to send to my MXChip IOT Device and when I click on "Select Serial Port" doesn't do nothing