microsoft/vscode-arduino #675: Unable to find the Arduino package path, please install the latest Arduino package for Devkit.

samirmj, 2018-10-19:
- Extension Name: vscode-arduino
- Extension Version: 0.2.22
- OS Version: Windows_NT x64 10.0.17134
- VSCode version: 1.28.2

 We have written the needed data into your clipboard. Please paste: I am getting error while Provisioning Azure services - "Unable to find the Arduino package path, please install the latest Arduino package for Devkit."

czgtest, 2018-11-05:
@samirmj  do you resolve this issue ?

closed 2018-11-26 by czgtest

noahsdonaldson, 2019-04-25:
I had this same error and in vscode it told me it couldn't access package_index.json.tmp in my app local directory. I just went in and changed the file extension name to .old and tried to install AZ3166 again and it worked just fine.