microsoft/vscode-arduino #678: Dependency error avr32/io.h

ueberBrot, 2018-10-24:
Hey,

i started today using VS Code for programming Arduino sketches. I opened a sketch that I programmed with the Arduino IDE. After opening the file i selected my board and the arduino.json + c_cpp_properties.json files were created in the .vscode folder.
With the help of the "Show Fixes" dropdown menu I  added the following to my c_cpp_properties.json file:

[image]

I still have 2 warnings:

`#include errors detected. Please update your includePath. IntelliSense features for this translation unit (D:\My\Path\to\the\script\myscript.ino) will be provided by the Tag Parser.`

`cannot open source file "avr32/io.h" (dependency of "SPI.h")`

When I peek the definition it clearly shows that its not defined.

[image]

However when I open the script with the Arduino IDE <SPI.h> works without any issues.
Why is that?

Hope you can help me.

Cheers

sslupsky, 2018-10-24:
I think you need to add some defines to the configuration.  Here is what I use for the MKRWAN1300:

            "defines": [
                "ARDUINO=100",
                "ARDUINO_ARCH_SAMD",
                "ARDUINO_SAMD_MKRWAN1300",
                "__SAMD21G18A__",
                "BOARD_ID_arduino_mkrwan1300",
                "_VARIANT_ARDUINO_MKRWAN1300_"
            ],

Since you are using AVR, your defines will be different but conceptually, you need to add them the have the conditional defines work.

Also, there is a file "includes.cache" in the build directory ... look inside that file and make sure all the includes listed in there are in your configuration.

ueberBrot, 2018-10-25:
I don't know what why but i deleted the .vscode folder and started the setup process again.
I now have a settings.json file with the parameter`"C_Cpp.intelliSenseEngine": "Tag Parser"`.

It seems the intellisense works now. maybe its because of that but i don't know that.

Nevertheless I will add defines for my board if i can figure out what to type in there :).

Where did you get your define values from?

closed 2018-11-23 by ueberBrot

ddsoyka, 2019-05-18:
I am also getting this error using an Arduino/Genuino Uno board on Windows 10.

`cannot open source file "avr32/io.h" (dependency of "SPI.h")`

This is my `c_cpp_properties.json` file:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\\\Program Files (x86)\\\\Arduino\\\\hardware\\\\arduino\\\\avr\\\\**",
                "C:\\\\Program Files (x86)\\\\Arduino\\\\hardware\\\\tools\\\\**",
                "C:\\\\Program Files (x86)\\\\Arduino\\\\tools\\\\**",
                "C:\\\\Program Files (x86)\\\\Arduino\\\\libraries\\\\**",
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.20.27508/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "defines": [
                "USBCON"
            ]
        }
    ],
    "version": 4
}
```

sslupsky, 2019-05-18:
@ueberBrot   Apologies, I missed your post back in October.  Did you figure this one out?

A good place to look is in the compiler and linker configurations.  Recently I found that the platform.txt and boards.txt files have relevant configurations in them and I think if you pick the defines from these they are the only ones you need.  You can also dig through the header files to figure out which ones you require.  This is what I originally did but it is a more tedious process.

Regarding the Tag Parser ... this is a sub optimal solution.  You really are not getting the benefit of IntelliSense when you are using the Tag Parser.  The difference is huge.

ueberBrot, 2019-05-23:
@sslupsky thank you for your replay. I never figured it out and I stopped looking why it didn't work because of time frame for my project.

> Regarding the Tag Parser ... this is a sub optimal solution. You really are not getting the benefit of IntelliSense when you are using the Tag Parser. The difference is huge.

Yeah exactly. Working with it I had many problems during the build process and that's why I switched back to the Arduino IDE.
Later I found an alternative with PlatformIO and I'm happy with it so far.

naigonakoii, 2020-07-22:
FYI, I know this issue is closed by I was able to get around this issue in my case by doing the following two things:

1. Add a direct path to the SPI library to the "includes" array, as follows:
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/src"

2. Modify the following two variables:
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe",
            "intelliSenseMode": "gcc-x86",

# 2 is needed as without it the SPI.h library has tons of IDE issues per MSVC. By doing this my project builds correctly and has no more squiggly lines issues.

Jon1105, 2020-11-09:
I believe this is a problem due to the `C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/keywords.txt` file
I'm not sure if this is for everyone, but line 2 of my file is `# Syntax Coloring Map SPI`, when I think it should be `# Syntax Coloring Map for SPI`.

All other libraries in the  `C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/` directory have line 2 of their keywords.txt at `# Syntax Coloring Map for{library name}`

I posted this [issue](https://github.com/arduino/Arduino/issues/10938) on the arduino github repo

per1234, 2020-11-09:
>I believe this is a problem due to the C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/keywords.txt file
I'm not sure if this is for everyone, but line 2 of my file is # Syntax Coloring Map SPI, when I think it should be # Syntax Coloring Map for SPI.

That's only a comment. It is indeed a typo, and I am not opposed to it being corrected (though I'm not in a position to merge PRs in the Arduino boards platform repositories), but a typo in a comment should never have any functional effect. If for some reason this project is parsing the comments in keywords.txt files (which I very much doubt), then that should be considered a faulty approach. There is no specification for any format for these comments. The library author is free to add anything they like in comments in these files, or no comments at all if they prefer.

tapir, 2021-01-23:
This problem still persists. Clean install of arduino and vscode extensions. First try with SPI library and got the exact same error that avr32/io.h can't be found. Why is this issue closed though?

hoekbrwr, 2021-02-05:
Having this problem too. Although Arduino IDE **is** compiling and VSCode also. All the dependency stuff is like Chinese to me! The dependency will probably be solved via another path. Which one????? stays a mystery.

ahbserv, 2021-02-22:
I'm having exactly the same problem on my windows 10 system. So I set it all up(arduino ide, vs code with arduino extension) on my macbook  air and the problem went away. I set c/cc++ configuration(on both systems) using the UI method and put all the includes (obtained by using verbose mode and looking at compiler call from arduino ide) in the compiler arguments . Perplexed but don't want to spend any more time on it (unless I have to).

Voclav, 2021-02-26:
I solved the problem by changing the order of the library locations in `c_cpp_properties.json`.

hoekbrwr, 2021-02-27:
Is there any logic in the arrangement of the libraries in
c_cpp_properties.json?

Voclav, 2021-02-27:
In my opinion, those libraries are searched in that order. And because they have the same names, the first one finds one that uses `avr/io.h` and not the one that uses `avr32/io.h`.

hoekbrwr, 2021-02-27:
I don't see any json file content. To what json file are you referring?

Voclav, 2021-02-27:
I changed the order to `"includePath":` and it helped me. But I program for ESP8266.

hoekbrwr, 2021-02-27:
I am at the same spot, but what comes first? I put in workspace first than the Arduino and than the "external" libs. Please show me a copy of your includePath branch of json file.

Voclav, 2021-02-27:
`{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\vacla\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\vacla\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\**",
                "C:\\Users\\vacla\\Documents\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "${workspaceFolder}/**"
            ],
            "defines": [
                "UNICODE",
                "_UNICODE",
                "ARDUINO=10805",
                "DDEBUG=false",
                "USBCON"
            ],
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64",
            "forcedInclude": []
        }
    ],
    "version": 4
}`

hoekbrwr, 2021-02-27:
Thanks! I will have a try to apply that for my libraries

beachmiles, 2021-03-16:
After a good couple hours pulling my hair out I finally got a c_cpp_properties.json config that works in Windows for a mega 2560 board. 
The ** was not finding all my subdirectories so I had to explicitly call out some directories... I suspect there is a bug in the handling of the space in the windows path ...

Now if I could only make this the default config for any arduino files I open up in Visual Studio Code... anyone know a way??

`{
    "configurations": [
        {
            "name": "Win32",
            "defines": [
                "ARDUINO=10800",
                "USBCON"
            ],
            "includePath": [
                "${workspaceRoot}",
                "${env:HOMEPATH}\\Documents\\Arduino\\libraries\\**",
                "${env:HOMEPATH}\\Documents\\Arduino\\hardware\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src\\",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SoftwareSerial\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "clang-x64",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe"
        }
    ],
    "version": 4
}`

ed-graham, 2022-04-21:
I think I understand what is happening, and will try to explain it.

**SOLUTION**

Open your `.vscode\c_cpp_properties.json` file and add the following line **at the top** of the list of `includePath` properties (or more precisely, before *any* other path containing `C:/Program Files (x86)/Arduino`, assuming that is your installation folder for Arduino under Windows):

```
C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/**
```

Then **restart VS Code** (since sometimes restarting IntelliSense on its own doesn't seem to be enough).  The dependency error should now disappear.  If other IntelliSense errors appear, ensure your `compilerPath` and `intelliSenseMode` settings are correct for Arduino development, as per @naigonakoii's comment above.

**EXPLANATION**

There are *several* files named `spi.h` beneath the `C:/Program Files (x86)/Arduino` folder (i.e., in a subfolder of it).  The Arduino IDE finds the right one; hence no build errors arise when using it.  However, VS Code IntelliSense finds the wrong one; hence the red squiggly lines.

If you compile your code in the Arduino IDE with verbose output (Ctrl+Comma > Show verbose output during compilation), near the bottom of the output window you should see a message like this:

```
...
Linking everything together...
...
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library Adafruit_INA219 at version 1.2.0 in folder: C:\Code\Arduino\libraries\Adafruit_INA219 
Using library Adafruit_BusIO at version 1.11.3 in folder: C:\Code\Arduino\libraries\Adafruit_BusIO 
--->                                                                                                                                                                             <---
---> Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI <---
--->                                                                                                                                                                             <---
```

In my case, it was the `Adafruit_BusIO` library that included `SPI.h`; specifically, the file

 `C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.h`.

Note that this file *does not include* the `avr32/io.h` file.  So that's the one VS Code should be picking up as well.

VS Code looks through the list of folders specified in the of `includePath` property in your `.vscode\c_cpp_properties.json` file **in order** until it finds a file named `SPI.h` - and unfortunately, it's *not* case-sensitive (on Windows, at least), so it will pick up files named `spi.h` as well.  A quick search for files named `spi.h` beneath my `C:/Program Files (x86)/Arduino` folder revealed the following, listed in alphabetical order:

```
C:\Program Files (x86)\Arduino\hardware\arduino\avr\firmwares\wifishield\wifiHD\src\SOFTWARE_FRAMEWORK\DRIVERS\SPI\spi.h
C:\Program Files (x86)\Arduino\hardware\arduino\avr\firmwares\wifishield\wifi_dnld\src\SOFTWARE_FRAMEWORK\DRIVERS\SPI\spi.h
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.h
C:\Program Files (x86)\Arduino\libraries\WiFi\extras\wifi_dnld\src\SOFTWARE_FRAMEWORK\DRIVERS\SPI\spi.h
C:\Program Files (x86)\Arduino\libraries\WiFi\extras\wifiHD\src\SOFTWARE_FRAMEWORK\DRIVERS\SPI\spi.h
```

Note that *all* the others contain the line

`#include <avr32/io.h>`

so if we hit any of these, we'll get the IntelliSense error.  My `c_cpp_properties.json` file previously looked like this (when I was seeing the error):

```
            "includePath": [
                "${workspaceFolder}/**",
                "C:/Program Files (x86)/Arduino/**",
                "C:/Code/Arduino/libraries/**"
```

Assuming VS Code is working through the `C:/Program Files (x86)/Arduino/**` directory tree (recursively) *in alphabetical order*, it will hit this one first:

`C:\Program Files (x86)\Arduino\hardware\arduino\avr\firmwares\wifishield\wifiHD\src\SOFTWARE_FRAMEWORK\DRIVERS\SPI\spi.h`

Therefore I modified it as suggested above to this:

```
            "includePath": [
                "${workspaceFolder}/**",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/**",
                "C:/Program Files (x86)/Arduino/**",
                "C:/Code/Arduino/libraries/**"
```

Unlike others, I saw no problems with VS Code resolving the recursion (`**`) with path names that include spaces.