microsoft/vscode-arduino #683: Open "sketch\bins\sketch\bins" verifying error

Jipolie01, 2018-11-03:
Hey,

So I've been working with the extension for a month now and everything worked fine. And then out of nowhere while verifying a file I got this error (I translated the words because my arduino is set in a different languages, so it might not be exactly the same):
```
Please see the build logs in Output path: c:\Users\jipga\Desktop\SOMNOX\Code\STM32-Main\bins
Loading configuration...
Loading packages...
Preparing boards...
Verify...
open sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\sketch\bins\preproc\ctags_target_for_gcc_minus_e.cpp: System can't find path
```
My arduino.json looks like this:
```
{
    "board": "Arduino_STM32:STM32F1:genericSTM32F103Z",
    "configuration": "device_variant=STM32F103ZC0,upload_method=STLinkMethod,cpu_speed=speed_72mhz,opt=osstd",
    "sketch": "STM32-Main.ino",
    "port": "COM7",
    "output": "./bins"
}
```
The actual path of the file that the compiler is trying to find is: sketch\bins\preproc\ctags_target_for_gcc_minu_e.cpp. 
I don't know what could have happend that changed this, everything was working fine. Even when reverting the code changes the error is still there. 
I changed the output directory multiple times, the error is still the same. 
Is there a setting that I can use to change this path? Or is this a different type of problem,  The arduino IDE has the same problem for me unfortunately.

czgtest, 2018-12-27:
this is an issue of Arduino IDE , https://github.com/arduino/Arduino/issues/8271

czgtest, 2019-01-21:
@Jipolie01  Arduino IDE has fixed this issue on https://github.com/arduino/Arduino/issues/8271 
Can you have a try ?

closed 2019-01-21 by czgtest

czgtest, 2019-01-21:
@Jipolie01  you can download  HOURLY BUILDS  https://www.arduino.cc/en/main/software
and update the "arduino.path" , then do verify action. 

[image]